wpilibsuite/GradleRIO #180: Should the system java.library.path be added to that of the simulate tasks?

amrelk, 2018-09-28:
I'm not sure about C++, so this issue may affect it also, but at least in java, when I run the simulate tasks, it only includes the JNI Extraction dir path in the `-Djava.library.path` argument. This means that libraries the are in my /usr/lib (e.g. libpathfinder.so) are not included. I've solved this so far by manually adding /usr/lib in the generated gradlerio_simulate.sh, but it looks like that won't be possible now that the task launches the process with the launch() method. Should /usr/lib (or whatever the windows equivalent is) be added to the ldpath variable in `edu.wpi.first.gradlerio.test.JavaSimulationTask` or am I just doing something painfully wrong?

JacisNonsense, 2018-09-29:
This is already done here: https://github.com/wpilibsuite/GradleRIO/blob/master/src/main/groovy/edu/wpi/first/gradlerio/test/JavaTestPlugin.groovy#L59

amrelk, 2018-09-30:
It doesn't look like it's done here, unless setting LD_LIBRARY_PATH in the other file solves that:
https://github.com/wpilibsuite/GradleRIO/blob/master/src/main/groovy/edu/wpi/first/gradlerio/test/JavaSimulationTask.groovy#L20