wpilibsuite/GradleRIO #184: Can't add custom task to deploy task.

y0shi, 2018-10-04:
Hi!  I'm trying to add my own custom task that would occur before the `deploy` task, but it seems that the deploy extension isn't allowing it.  (Disclaimer:  I use gradle, but I usually don't define custom tasks.)  

Steps to recreate:

I defined a task in the `build.gradle` file:

``` groovy
task myTask << {
    println('Well hello there!')
}
```

Immediately below that,  I add a `dependsOn()` to the `compileJava` task.  Everything is ok:

``` groovy
// This works.
project.compileJava.dependsOn(myTask)
```

But if I do the same with the deploy task, it fails:
``` groovy
// This fails 
project.deploy.dependsOn(myTask)
```

The build error I get is:
```
> Could not find method dependsOn() for arguments [task ':myTask'] on object of type jaci.gradle.deploy.DeployExtension.
```

ThadHouse, 2018-10-04:
Project.deploy gets the deploy extension variable, not the deploy task. In order to get that, you should use

`tasks.named('deploy').depends on(myTask)`

y0shi, 2018-10-04:
Hmm... maybe I'm still doing some thing wrong.  I tried the following combinations and got these errors respectively:

```
tasks.named('deploy').depends on(myTask)

> Could not find method on() for arguments [task ':myTask'] on root project 'Robot_2018_VS' of type org.gradle.api.Project.

tasks.named('deploy').dependsOn(myTask)

> Could not find method dependsOn() for arguments [task ':myTask'] on provider(task deploy, class org.gradle.api.DefaultTask) of type org.gradle.api.internal.tasks.DefaultTaskContainer$TaskCreatingProvider_Decorated.

tasks.named('deploy').dependsOn myTask

> Could not find method dependsOn() for arguments [task ':myTask'] on provider(task deploy, class org.gradle.api.DefaultTask) of type org.gradle.api.internal.tasks.DefaultTaskContainer$TaskCreatingProvider_Decorated.

tasks.named('deploy') {
    dependsOn myTask
}

> Could not find method named() for arguments [deploy, build_a4173o9n9vhxwe6iuea413g94$_run_closure4@77e92eaf] on task setof type org.gradle.api.internal.tasks.DefaultTaskContainer.
```

y0shi, 2018-10-04:
Awesome!  Thank you for the help!

closed 2018-10-04 by y0shi