wpilibsuite/GradleRIO #186: AndyMark LED Controller

AustinShalit, 2018-10-05:
_From @tcroots78 on September 27, 2018 1:33_

Trying to use our code from last year that has https://www.andymark.com/product-p/am-3847.htm and we get errors in the main LightDrive.h that is provided (the one they provide) in the downloads and it has instructions for installing it.

Instead of putting it in the includes, we had just dropped it in with our Robot code last year and that seemed to work. Although I think I remember eclipse complaining about some stuff...but it compiled and worked...and was during build season...so we never quite prioritized it 😲 

The instructions also mention putting files into the wpilib/user/cpp/lib directory...which I think we needed to do, but not sure how to do that now with these pulling in through gradle.

_Copied from original issue: wpilibsuite/vscode-wpilib#92_

AustinShalit, 2018-10-05:
_From @tcroots78 on October 4, 2018 21:38_

Played around a bit trying to see what includes and libs were being accessed

I was able to get IntelliSense working by putting the LightDrive.h and LightDriveCAN.h files in 
/Library/Developer/CommandLineTools/usr/include/c++/v1

but no joy in getting the build to work tried putting the libMachLib.a in either:
/usr/local/arm-frc-linux-gnueabi/lib
/usr/local/arm-frc-linux-gnueabi/usr/lib
(and .h in the related include)

AustinShalit, 2018-10-05:
_From @ThadHouse on October 4, 2018 21:39_

Thats definitely not where you want to put either of those. Basically to make things work you're going to need to add things to your build.gradle file to manually link that binary. It might be better to move this over to the GradleRIO repo as this more pertains to that. GradleRIO handles the build.

AustinShalit, 2018-10-05:
I am going to move this issue over to GradleRIO because it is a GradleRIO documentation issue.

tcroots78, 2018-10-06:
Can you elaborate on what needs to be done to manually link that binary in the build.gradle?

JacisNonsense, 2018-10-08:
To bind a library in C++, you have to put the .zip file in your project directory and put the following in your build.gradle:
```gradle
model {
    libraries {
        mylib(NativeLib) {
            headerDirs << 'cpp'                         // Directories for headers
            staticMatchers << '**/*.a'                      // Static Libraries to be linked at compile time
            
            file = project.file("mylib.zip")                // Load from filesystem instead. Can be given a zip or a directory.
        }
    }
}
```

Then you can put `mylib` in the `useLibrary` declaration.

sleepisfortheweak, 2018-10-10:
Using your example, I get:
`No such property: NativeLib for class: org.gradle.model.ModelMap`

I also tried changing NativeLib to NativeLibrary, but get:
`Cannot create 'libraries.mylib' with type 'org.gradle.nativeplatform.NativeLibrary' as this is not a subtype of 'jaci.gradle.nativedeps.BaseLibSpec'.`

I've attached my build.gradle file. Am I missing something here? Thanks.

[build.gradle.txt](https://github.com/wpilibsuite/GradleRIO/files/2463143/build.gradle.txt)

JacisNonsense, 2018-10-10:
You need to fully quantify the NativeLib class, or put an import in your build.gradle. NativeLib sits in the `jaci.gradle.nativedeps` package

sleepisfortheweak, 2018-10-11:
Thanks for the reply. I have that working now, but just can't get this to build. See my attached build.gradle. If I add 'mylib' to useLibrary(), it says:

`Could not determine the dependencies of task ':linkFrcUserProgramExecutable'. > Cannot find delegated dependency: mylib for binary: executable 'frcUserProgram:executable'`

However, if I don't add 'mylib' to useLibrary(), I just get the expected "undefined reference" error from gcc. No matter what I put into the 'headerDirs' or 'file' parameters, I don't get any errors or feedback. I only get errors if I intentionally screw up the parameter names themselves. Either way it doesn't seem to 'see' mylib.

Am I defining my paths correctly? I assume all paths are relative to my project directory. Is 'file' supposed to point to the .a file, its directory, or something else?
[build.gradle.txt](https://github.com/wpilibsuite/GradleRIO/files/2467246/build.gradle.txt)

sleepisfortheweak, 2018-10-11:
This probably isn't the _best_ way, but I do have it working by 'cheating' and putting the correct args into the linker. I just add:

`linker.args('-LC:/FRC_Projects/Test_CPP/lib','-lMACHLib')`

...To my build.gradle under the `cppCompiler.args` line - using your own -L path to your libMACHLiba.a file of course.

JacisNonsense, 2018-10-11:
`file =` should point to your zip file, not the .a

sleepisfortheweak, 2018-10-11:
What is the structure of the ZIP file? I just put the .a in it, but I still got the "Could not determine dependencies" error.

JacisNonsense, 2018-10-11:
It's the zip file you download from andymark

sleepisfortheweak, 2018-10-11:
Ah, that's why headerDir is 'cpp' - to match the structure in the ZIP file? Taking that into account, I still get the same error. I know it 'sees' the file, because if I alter the path, I get a different error, but when it's correct, I get the "Could not determine dependencies error".

JacisNonsense, 2018-10-11:
Can you post a stacktrace of your build error? `./gradlew build --stacktrace`. Also let us know what version of GradleRIO you are using

sleepisfortheweak, 2018-10-11:
Stacktrace attached. GradleRIO is 2019.0.0-alpha-3.
[stacktrace.txt](https://github.com/wpilibsuite/GradleRIO/files/2467558/stacktrace.txt)

JacisNonsense, 2018-10-11:
Try updating GradleRIO to `2019.0.0-beta0-pre3` and see if that results in a fix

sleepisfortheweak, 2018-10-12:
When I update to beta0-pre3, I get the 'Could not determine dependencies' error, but for _CTRE_ libs this time. When I remove all deps except for "wpilib" and "mylib", I get the same error for mylib.

JacisNonsense, 2018-10-12:
Whoops, that was my bad. Revert to alpha-3 and use the following structure instead (forgot `targetPlatforms`). 

```
mylib(jaci.gradle.nativedeps.NativeLib) {
    headerDirs << 'cpp'            
    staticMatchers << '**/*.a'
    
    file = project.file("lib/LDRV-12_Examples.zip")
    targetPlatforms << 'roborio'
}
```

As for the CTRE error, that's because we're not bundling vendor libraries in by default for 2019, instead vendors will be providing that functionality

sleepisfortheweak, 2018-10-12:
BUILD SUCCESSFUL

JacisNonsense, 2018-10-12:
For vendors, we're getting them to distribute their lib information through some .json files that GradleRIO will be able to read. For that to work, you'll first need to publish your library on maven. We're still working on writing all this up, so hang tight

JacobCover, 2019-02-08:
Has anything changed with this yet? I'm trying to use the lightdrive this year and am not really sure where too start. Using Java, not cpp. Thanks!

sleepisfortheweak, 2019-02-12:
The INSTALL.txt file has been updated in the Examples link on the AndyMark product page. It explains where you need to put files and when lines in build.gradle to modify.

However, this is not the _official_ way of adding support. We're trying to get more information on how to produce the required JSON files for Gradle to pickup the dependencies automatically.

JacobCover, 2019-02-12:
Ok great, I'll look into that today. You saw the CTRE json right? You could
use that as a starting point.

JacobCover, 2019-02-13:
I tried getting the LightDrive to work, but I can't get it to work on the CAN bus since it seems to only have 2 CAN inputs. There also seems to be no way in the library to use the serial control scheme. If someone could help me get either (preferably CAN) working, that would be great!

sleepisfortheweak, 2019-02-13:
I don't completely understand what you mean by only having 2 CAN inputs, but you just have to attach the wires like you would on the PDP, PCM, or other CAN device. I put it on the end of the line usually, but if you want to pass-through it, you could twist the in and out wires together and insert them. Once connected and initialized, you should see the onboard LED go purple. That indicates a valid CAN signal.

JacobCover, 2019-02-13:
Oh ok, I didn't realize you could twist to can wires together and use one
port, my team hasn't done much CAN beyond the CTRE stuff. I'll test this in
a few hours, thanks!

sleepisfortheweak, 2019-02-14:
We have a VS Code-compatible vendor library, but remaining issues seem to be related to our library being compiled against an older version of WPILib. This should be corrected today. In the meantime you may see your robot code crash with ClassNotFoundException on launch if including the LightDrive library.

sleepisfortheweak, 2019-02-15:
OK, Java should now be working with VS Code using the built-in Vendor Libraries Function. C++ will follow shortly with the same procedure. The INSTALL.txt instructions file has been updated on the product page, but you can also follow the instructions below:

1. Open the Command Palette and type "Manage Vendor Libraries"
2. Then click "Install New Libraries (online)"
3. Enter the following URL: https://mach-engineering.com/products/maven/com/mach/LightDrive/LightDrive.json
4. Press Enter and the library is added.

JacobCover, 2019-02-17:
@sleepisfortheweak There is a problem with your provided JSON, it fails to build. To fix it, just change the maven URL from 
```    "mavenUrls": [
        "https://mach-engineering.com/products/maven/com/mach/LightDrive"
    ],```
to 
```    "mavenUrls": [
        "https://mach-engineering.com/products/maven/"
    ],```.

For some reason, Gradle appends `com/mach/LightDrive` to the end of the maven URL automatically. Seen the attaches pastebin for the working LightDrive.json.
https://pastebin.com/GM3tzTnH

sleepisfortheweak, 2019-02-17:
Thanks for your reply. I uploaded the fixed JSON. It always works on my machine somehow - even when starting a project from scratch. I've had at least one customer contact me with related build issues, so thanks for potentially saving a lot of time.

I have javadocs, but am not sure how to include it in my package. VS Code doesn't seem to pick them up automatically like it does with -source files. Is there a supported way to provide docs?

ThadHouse, 2019-02-17:
Upload the javadoc jar with -javadoc as a sufficient before the .jar to the same folder your current jar is in. It will automatically pick them up then.

sleepisfortheweak, 2019-02-17:
OK, that is uploaded. How do I confirm it got them? Can't seem to find the cache it puts the JARs in... Not seeing hover tips in VS Code, but I guess that could be another issue.

sleepisfortheweak, 2019-02-17:
Found it in my .m2 folder once I looked close enough. However only my library JAR appears there. It doesn't seem to pick up -javadoc or -source JARs from the repo. I even tried being 'clever' and adding them to the .m2 folder manually under their own SHA-1 hashes like the others, but VS Code never seems to see them. I was only able to get -sources to work earlier when I installed the library using the local method under the frc2019 public folder.

ThadHouse, 2019-02-17:
You might actually need to have the -sources jar published in addition to the -javadoc jar. Not sure if intellisense will try to pick up docs without sources.

JacisNonsense, 2019-02-18:
I'm going to close this since this seems to be a vendor-side conversation at the moment, not a bug in GradleRIO

closed 2019-02-18 by JacisNonsense