wpilibsuite/GradleRIO #211: Java remote debug broken

PeterJohnson, 2018-10-19:
At least on Java 11, the JDWP by default only binds to localhost, not all addresses.  This makes remote debug impossible.  To fix this, the JDWP address option must be passed as `0.0.0.0:8348`, not just `8348`, e.g. this works:

`/usr/local/frc/JRE/bin/java -Djava.library.path=/usr/local/frc/lib/ -XX:+UsePerfData -agentlib:jdwp=transport=dt_socket,address=0.0.0.0:8348,server=y,suspend=y -jar /home/lvuser/test-java-project.jar`

but this does not:

`/usr/local/frc/JRE/bin/java -Djava.library.path=/usr/local/frc/lib/ -XX:+UsePerfData -agentlib:jdwp=transport=dt_socket,address=8348,server=y,suspend=y -jar /home/lvuser/test-java-project.jar`

JLLeitschuh, 2018-10-19:
Out of curiosity, why are both the java debugger and the GDB debugger both using port `8348`? What if you want to debug in the JNI bindings for a Java application?

JLLeitschuh, 2018-10-19:
These are the lines requiring a fix:
https://github.com/wpilibsuite/GradleRIO/blob/1c387f5f4fdfc42b89c96302f5d6dee308e404ce/src/main/groovy/edu/wpi/first/gradlerio/frc/FRCJavaArtifact.groovy#L33-L34

PeterJohnson, 2018-10-19:
I agree the GDB port should be changed.

commit c29c9f71 (2018-11-02): Fix java debugging and change port (#212)

Debugging needs to be addressed to 0.0.0.0 in order to allow remote connections

Debugging port shouldn't match between C++ and Java in order to allow dual mode debugging in the future

Closes #211
closed 2018-11-02 by JacisNonsense