wpilibsuite/GradleRIO #213: NoSuchMethodError when running gradlew

coolioasjulio, 2018-10-26:
When I run any gradle command, (`gradlew`, `gradlew idea`, `gradlew wpi`, etc), the operation fails with a `java.lang.NoSuchMethodError: org.gradle.api.plugins.PluginAware.getPluginManager()Lorg/gradle/api/plugins/PluginManager;`.

This is the output of running `gradlew --stacktrace`: https://pastebin.com/hDj3HK8q

This is my `build.gradle`: https://pastebin.com/Ftd3Zh4c

I downloaded the release 3.0.0 (GradleRIO Universal) from the github page and extracted that into my repo. The original `build.gradle` didn't work, so I had to change the repo location and gradleRIO dependency information.

Doing some research, it seems like my gradle wrapper may be out of date so I changed it from 2.2.1 to 2.6, but that caused a lot more errors in different places. And since 2.2.1 was the version specified in the release, that leads me to believe that it was intentional.

I don't have a lot of experience with gradle, so it's possible I'm using it completely incorrectly. What am I doing wrong?

ThadHouse, 2018-10-26:
That version of GradleRio requires gradle 4.4. The template in releases is a few years old.

coolioasjulio, 2018-10-26:
That fixed it, thanks!

Upgrading the gradle version to 4.7 fixed all issues.

closed 2018-10-26 by coolioasjulio