wpilibsuite/GradleRIO #214: Build fails with error message.

tm1287, 2018-10-26:
When I run the "./gradlew" 'command on the gradle files included in the Quickstart release, the build fails and this error message appears.

FAILURE: Build failed with an exception.
***
What went wrong:
A problem occurred configuring root project 'GradleRIO'.
   Could not resolve all dependencies for configuration ':classpath'.
   Could not find any version that matches jaci.openrio.gradle:GradleRIO:+.
     Searched in the following locations:
         https://repo1.maven.org/maven2/jaci/openrio/gradle/GradleRIO/maven-metadata.xml
         https://repo1.maven.org/maven2/jaci/openrio/gradle/GradleRIO/
         http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/maven-metadata.xml
         http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/
     Required by:
         :GradleRIO:unspecified

Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 2 mins 5.043 secs
***

Running ./gradlew --stacktrace yields

https://pastebin.com/zgL8Yede
***
Running ./gradlew --info yields

Starting Build
Settings evaluated using empty settings script.
Projects loaded. Root project using build file 'C:\Users\Tejas\Downloads\GradleRIO\GradleRIO\build.gradle'.
Included projects: [root project 'GradleRIO']
Evaluating root project 'GradleRIO' using build file 'C:\Users\Tejas\Downloads\GradleRIO\GradleRIO\build.gradle'.
Resource missing. [HTTP GET: https://repo1.maven.org/maven2/jaci/openrio/gradle/GradleRIO/maven-metadata.xml]
Resource missing. [HTTP GET: https://repo1.maven.org/maven2/jaci/openrio/gradle/GradleRIO/]
Resource missing. [HTTP GET: http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/maven-metadata.xml]
Resource missing. [HTTP GET: http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/]

FAILURE: Build failed with an exception.

What went wrong:
A problem occurred configuring root project 'GradleRIO'.
> Could not resolve all dependencies for configuration ':classpath'.
   > Could not find any version that matches jaci.openrio.gradle:GradleRIO:+.
     Searched in the following locations:
         https://repo1.maven.org/maven2/jaci/openrio/gradle/GradleRIO/maven-metadata.xml
         https://repo1.maven.org/maven2/jaci/openrio/gradle/GradleRIO/
         http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/maven-metadata.xml
         http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/
     Required by:
         :GradleRIO:unspecified

Try:
Run with --stacktrace option to get the stack trace. Run with --debug option to get more log output.

BUILD FAILED

JacisNonsense, 2018-10-26:
Looks like you're missing `version: ` in your build.gradle.
Could you send your build.gradle file?

tm1287, 2018-10-26:
Here it is: https://pastebin.com/ci02c0mh

JacisNonsense, 2018-10-26:
What version of GradleRIO are you using? This template is super old. You should be using the Alpha versions: https://wpilib.screenstepslive.com/s/currentCS/m/79833/l/943086-alpha-test-info

tm1287, 2018-10-26:
I was using the Quickstart release on the releases section of the repository. I release now that it was released in 2015.

JacisNonsense, 2018-10-26:
It's explained in the above screensteps link, a few pages thru

closed 2018-10-26 by JacisNonsense