wpilibsuite/GradleRIO #216: Add warning if deploying without libs being downloaded

JacisNonsense, 2018-10-28:
It should prompt the user to first run` ./gradlew build` or trigger build from the IDE

JLLeitschuh, 2018-10-29:
Why not just make the deploy depend upon `build`? That way this never happens and nobody ends up with stale artifacts?

JacisNonsense, 2018-10-29:
Deploy is called when connected to robot wifi. Deploy already depends on build tasks for each artifacts. The wifi issue is what I'm trying to fix - since libs can't download when not connected to the internet

JLLeitschuh, 2018-10-29:
Fascinating. Wouldn't this be an error case then? Not a warning? I don't understand how the deploy could succeed if the dependent libraries weren't downloaded.

ThadHouse, 2018-10-29:
Yes, but it would be nice to have an actual message, rather then just dependency not found. Our people are REALLY bad at even the most simple of problem solving.

JLLeitschuh, 2018-10-29:
100% agreed.

JacisNonsense, 2018-10-30:
By warning I do mean error case - usually the build would fail, but if I can hook on build fail and check that some dependencies are, indeed, missing, I can give a more idiomatic error message.

If there are artifacts that aren't downloaded, but the deploy still succeeds, we shouldn't be the ones causing the failure

JacisNonsense, 2018-10-30:
Wasn't meant to close that, whoops

commit 4e89c503 (2018-11-05): Add Build Failure checkers to give plain-english error messages for common errors. Closes #216
closed 2018-11-05 by JacisNonsense