wpilibsuite/GradleRIO #220: Building issue

vieiraPatrick, 2018-11-07:
Every time I try to build my code the line "import edu.wpi.first.wpilibj.IterativeRobot;" disappears and I get an error.

PeterJohnson, 2018-11-07:
Is this with the beta?  Are you using vscode?  Is this on Windows or some other platform?  Does the line disappear when you save, or when you build/deploy?  Can you edit in a text editor outside of vscode and confirm that the line is present there, and manually build in a command prompt?

vieiraPatrick, 2018-11-07:
Hi Peter,

It is with the Alpha version of the VS code, because I don't know how to
get access to download the Beta version. I am using windows 10. Every time
I try to save or build/deploy this happens, I also noticed that it is not
only to that one specific line, it happens when I used XboxController as
well. At first when I was using the example Getting Started it was building
normally, then after any changes I do this starts to happen. If I use the
same code with Eclipse it works.

*Patrick R. Vieira*
Southeast Career and Technical Academy
*Robotics Teacher*

"In the end, we only regret the chances we didn’t take."
*- Lewis Carroll*

PeterJohnson, 2018-11-07:
If you go to Preferences | Settings, and search for "organize", is "Organize Imports" checked or unchecked?  Can you upload a copy of your settings.json?  (you can get to it via Preferences | Settings and then clicking the "..." icon).

vieiraPatrick, 2018-11-07:
The option Organize Imports is not checked. I tried checking it and it
didn't change anything, so I just switched back. When I open the
settings.json it shows the following in the User Settings:

{
"java.errors.incompleteClasspath.severity": "ignore",
"wpilib.alwaysAskForTeamNumber": true
}

*Patrick R. Vieira*
Southeast Career and Technical Academy
*Robotics Teacher*

"In the end, we only regret the chances we didn’t take."
*- Lewis Carroll*

vieiraPatrick, 2018-11-07:
Workspace settings shows the following:

{
"java.configuration.updateBuildConfiguration": "automatic",
"files.exclude": {
"**/.git": true,
"**/.svn": true,
"**/.hg": true,
"**/CVS": true,
"**/.DS_Store": true,
"bin/": true,
".classpath": true,
".project": true
},
"wpilib.online": true
}

*Patrick R. Vieira*
Southeast Career and Technical Academy
*Robotics Teacher*

"In the end, we only regret the chances we didn’t take."
*- Lewis Carroll*

vieiraPatrick, 2018-11-08:
Does it look like I could try changing configuration or something?

*Patrick R. Vieira*
Southeast Career and Technical Academy
*Robotics Teacher*

"In the end, we only regret the chances we didn’t take."
*- Lewis Carroll*

JacisNonsense, 2018-12-30:
I'm cleaning out old issues and I can't quite tell what's wrong with this one. @vieiraPatrick , if you're still having issues, reopen this issue, but for now I'm going to assume it's resolved

closed 2018-12-30 by JacisNonsense