wpilibsuite/GradleRIO #222: Instructions to use GradleRIO without VSCode?

19lmyers, 2018-11-10:
The 2019 specification seems to have changed a bit, so I'd like to know if there's anything important I'll need to do to use GradleRIO in 2019 without having VSCode or the WPILib plugin installed.

I noticed the quickstart instructions are gone, so I looked at the example Java project and found quite a few references to methods that only make sense when using VSCode.

JacisNonsense, 2018-11-11:
Documentation for all aspects of the 2019 changes are still on the TODO list, so this is an upcoming solution

roshanr10, 2018-12-03:
Has the old `./gradlew idea` command been removed in the interim?

I'm looking to update our existing base project in IntelliJ to the 2019 WPILib, and I'm unable to find an easy way to do so.

19lmyers, 2018-12-04:
As someone else who's going to be doing this in January, I highly suspect
it comes down to including the idea plugin in the build.gradle.

JacisNonsense, 2018-12-05:
The `idea` plugin isn't applied by default in the new templates, you have to add 
```
plugins {
  id 'idea'
}
```
to the `plugins{}` block at the top of your build.gradle.

19lmyers, 2018-12-06:
But it still works if I do?

roshanr10, 2018-12-10:
Worked like a charm, thanks Jaci!

JacisNonsense, 2018-12-30:
This should now be fixed in the README

closed 2018-12-30 by JacisNonsense