wpilibsuite/GradleRIO #241: Which JDK Version Should Be Used To Compile GradleRIO Projects?

TheArchNemesis, 2018-12-07:
Hello, I Am Simon From Team 2212.
One of my team's projects is a library that contains utility classes the extend WPILib. Recently, we tried to make this library work with gradle. and while it seems to mostly work fine we ran into lots of issues with code that was built using different versions of the JDK. For example, the third-party opencv package is compiled for java 11, but the rest of WPILib is compiled for java 8, and as I understand the code that we upload onto the RoboRIO should target java 8 as well.

PeterJohnson, 2018-12-07:
For the season, everything will be Java 11, so that’s what the beta is using and what our latest published opencv packages are compiled with, etc.  The “alpha test” release uses Java 8, but if you want opencv you have to target the alpha test release opencv dependencies, not the latest ones we have published.

TheArchNemesis, 2018-12-07:
Thank you very much!

closed 2018-12-07 by TheArchNemesis