wpilibsuite/GradleRIO #247: Cannot compile projects with a path containing '

nathanmutin, 2018-12-09:
When I try to create a project in a folder containing an apostrophe, this project failed to build.
For example, a project named "test" in a folder named "Robo'Lyon" give me this output :
```
> Task :compileFrcUserProgramExecutableFrcUserProgramCpp FAILED
C:\Users\Nathan\Documents\Robo'Lyon\test\src\main\cpp\Robot.cpp:8:19: fatal error: Robot.h: No such file or directory
```

ThadHouse, 2018-12-09:
I'm not actually surprised that has an issue, but that might be an issue internal to Gradle itself. Can you send a few files?

The build.gradle and src folders.
Inside the build folder, there is a folder called tmp. Can you send everything from that except for the expandedArchives folder.

ThadHouse, 2018-12-10:
@nathanmutin out of curiousity, did this work back in eclipse?

ThadHouse, 2018-12-10:
Through more testing, this is an issue deep inside Gradle that we will likely not be able to fix for the season. So for now, move the project outside of the folder with the special character. We will keep this open in hope of an internal Gradle fix at some point.

nathanmutin, 2018-12-10:
Maybe I should post the issue on [vscode-wpilib](https://github.com/wpilibsuite/vscode-wpilib) so that the project creator refuse paths containing apostrophe ?

ThadHouse, 2018-12-10:
Sure, you can do that and I'll get to it for the season.

sciencewhiz, 2021-10-31:
Gradle now appears to work with apostrophes in the path, but deploying doesn't update the robotCommand (even though deploy appears successful

```
echo '/usr/local/frc/JRE/bin/java -XX:+UseConcMarkSweepGC -Djava.library.path=/usr/local/frc/third-party/lib -Djava.lang.invoke.stringConcat=BC_SB  -jar "/home/lvuser/2022'NewCommandJava.jar" ' > /home/lvuser/robotCommand
```

The apostrophe probably needs to be escaped