wpilibsuite/GradleRIO #269: Issue with deploying pathfinder

rtrahms, 2018-12-24:
Hi!
I am using 2019 beta 4 of libraries and tools, and have successfully built and deployed code using CTRE and NavX third party libraries.  When I attempt to install pathfinder third party library, it doesn't go so well.

First, from a project that has CTRE and navx vendordeps already set up, I attempt to 'manage vendor libraries/install 
new library (online)', and specify the following URL: https://dev.imjac.in/maven/jaci/pathfinder/PathfinderOLD-latest.json.  

No messages, nothing.  And no PathfinderOLD.json shows up in vendordeps folder (also, my local maven directory has only jaci/EmbeddedTools, no jaci/pathfinder folder which I would expect).

Next, if I grab the json directly from the  jaci maven site and drop it into my local frc-2019/vendordeps folder, I am able to 'manage vendor libraries/install new libraries (offline)', specifying PathfinderOld.  Everything referencing jaci.pathfinder in my java code now builds.  Still trying to determine the location of the pathfinder jars, but under 
the hood the following command line is being successfully run:
> Executing task: gradlew build  -Dorg.gradle.java.home="C:\Users\Public\frc2019\jdk" <

The ./gradlew downloadAll --stacktrace yields this:

PS C:\DATA\Git_Repositories\FRC-2019\ChillySwervePathfinder> ./gradlew downloadAll --stacktrace

> Configure project :
NOTE: You are using a BETA version of GradleRIO, designed for the 2019 Season!
This release requires the 2019 RoboRIO Image, and may be unstable. Do not use this for the official competition season.
If you encounter any issues and/or bugs, please report them to https://github.com/wpilibsuite/GradleRIO

> Task :downloadAll FAILED
Resolving: annotationProcessor
Resolving: apiElements

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':downloadAll'.
> Resolving configuration 'apiElements' directly is not allowed

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':downloadAll'.
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.
java:96)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:
65)
        at org.gradle.api.internal.tasks.execution.ActionEventFiringTaskExecuter.execute(ActionEventFiringTaskExecuter.
java:44)
        at org.gradle.api.internal.tasks.execution.TimeoutTaskExecuter.execute(TimeoutTaskExecuter.java:53)
        at org.gradle.api.internal.tasks.execution.SnapshotAfterExecutionTaskExecuter.
execute(SnapshotAfterExecutionTaskExecuter.java:38)
        at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.
execute(OutputDirectoryCreatingTaskExecuter.java:51)
        at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:59)
        at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.
execute(ResolveTaskOutputCachingStateExecuter.java:49)
        at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:61)
        at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.
execute(SkipEmptySourceFilesTaskExecuter.java:101)
        at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:
91)
        at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter
.java:44)
        at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.
execute(ResolveTaskArtifactStateTaskExecuter.java:62)
        at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.
java:55)
        at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:54)
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:
35)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.run(EventFiringTaskExecuter.java:49)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.
execute(DefaultBuildOperationExecutor.java:301)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.
execute(DefaultBuildOperationExecutor.java:293)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:175)
[... 308 lines cut ...]
        at org.apache.http.impl.execchain.MainClientExec.establishRoute(MainClientExec.java:394)
        at org.apache.http.impl.execchain.MainClientExec.execute(MainClientExec.java:237)
        at org.apache.http.impl.execchain.ProtocolExec.execute(ProtocolExec.java:185)
        at org.apache.http.impl.execchain.RetryExec.execute(RetryExec.java:89)
        at org.apache.http.impl.execchain.RedirectExec.execute(RedirectExec.java:110)
        at org.apache.http.impl.client.InternalHttpClient.doExecute(InternalHttpClient.java:185)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83)
        at org.gradle.internal.resource.transport.http.HttpClientHelper.performHttpRequest(HttpClientHelper.java:132)
        at org.gradle.internal.resource.transport.http.HttpClientHelper.performHttpRequest(HttpClientHelper.java:109)
        at org.gradle.internal.resource.transport.http.HttpClientHelper.executeGetOrHead(HttpClientHelper.java:98)
        at org.gradle.internal.resource.transport.http.HttpClientHelper.performRequest(HttpClientHelper.java:89)
        ... 52 more

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.0/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 4s
9 actionable tasks: 7 executed, 2 up-to-date
GradleRIO detected this build failed due to missing dependencies!
Try again with `./gradlew downloadAll` whilst connected to the internet (not the robot!)
If the error persists, ensure you are not behind a firewall / proxy server (common in schools)
PS C:\DATA\Git_Repositories\FRC-2019\ChillySwervePathfinder>

ThadHouse, 2018-12-24:
The manage vendor libraries online issue we know about, and it has been fixed in master for the vscode repo.

rtrahms, 2018-12-24:
Fixed in the v2019.1.1-beta-4 release, or do I need to clone and build the vscode-wpilib repo?
Thanks,
Rob

ThadHouse, 2018-12-24:
#272 fixes the downloadAll issue, but both of these issues wouldn't cause pathfinder to not deploy. What errors do you get when you actually try to deploy?

ThadHouse, 2018-12-24:
Ah, ok. Try adding the following to your build.gradle, and then running `./gradlew getAllArtifacts`. The code is basically just copying the task with the fix to make it work properly.

```
task getAllArtifacts() {
    doLast {
        project.configurations.each { Configuration conf ->
            if(!conf.canBeResolved) {
                return
            }
            println("Resolving: " + conf.getName())
            conf.resolvedConfiguration.resolvedArtifacts.each { ResolvedArtifact art ->
                art.file    // Needed to trigger download
            }
        }
    }
}
```

rtrahms, 2018-12-24:
That did the trick!  
./gradlew getAllArtifacts succeeded, and subsequent deploys to roborio also succeeded.
Thanks for the help.
Rob

ThadHouse, 2018-12-24:
Cool. The release will have downloadAll working, but that's the way to manually do it