wpilibsuite/GradleRIO #282: Creating and referencing roborio platform library projects

sjcbulldog, 2019-01-04:
With the alpha version of GradleRIO we were able to have static library projects that contains the code that was reused year to year (NativeLibrarySpec).  However, with the 2019.1.1 release we are not able to make this work.  The library projects seem to build, but we cannot reference them from the main robot project (NativeExecutableSpec).  Are there instructions anywhere on how to do this with the 2019.1.1 release of GrableRIO?

ThadHouse, 2019-01-04:
That should work the exact same way. What errors are you getting? Intellisense errors in VS Code, or actual build/deploy errors?

sjcbulldog, 2019-01-04:
To be clear, in the alpha version we added the following to the sources section:

				lib library: "xerobase", linkage: "static"
				lib library: "xeromath", linkage: "static"
				lib library: "xeromisc", linkage: "static"     

In the actual libraries sections, we had exported headers like shown in the snippet below.

			sources.cpp {
			    source {
				    srcDirs = ["xerobase"]
					include "**/*.cpp"
				}
				exportedHeaders {
				    srcDirs = ["xerobase"]
					include "*.h", "**/*.h"
				}

				lib library: "xeromath", linkage: "static"
				lib library: "xeromisc", linkage: "static"
			}

The problem we are seeing is the compilation of the top most project, does not find the include files from the library.  If I  build only the static library, the library builds just fine, but when I build the top most project, I get an error about not finding include files that exist in the libraries.

Thansk

ThadHouse, 2019-01-04:
Hmm. That should work, nothing there should have changed. Can I see the project somewhere where I can look.?

sjcbulldog, 2019-01-04:
Ok, let me wipe the egg from my face.  While typing the above, I sent back to verify what I told you and one subdirectory did not get copied over.  My fault completely.  Apologize for occupying your time :).

closed 2019-01-04 by sjcbulldog

ThadHouse, 2019-01-04:
No worries, glad its working.