wpilibsuite/GradleRIO #284: Question: Dependency ordering

sjcbulldog, 2019-01-05:
As mentioned in previous issues we have our code divided into libraries.  Everything seems to build now that I understand the system a little better, but I am having a problem with linking.  The issue is that I have our libraries that are part of our robot code.  In addition there are the libraries that come from the vendor dependencies.  The problem is that our libraries come last in the link command line, after the vendor libraries.  However our library code has a dependency on the vendor libraries.  I suspect this will be the case for many teams.  Is there a way to change this link order?  Of it not can the link command line wrap all of the libraries in a "-Wl,--start-group", "-Wl,--end-group" pair so that the libraries are traversed until all dependencies are satisfied?

ThadHouse, 2019-01-05:
The order happens in the declaration order, so if you put your library references before the useLibrary and useVendorLibraries calls, yours should show up first.

sjcbulldog, 2019-01-05:
The project is as shown below.  The only thing I do to get the vendor libraries is the wpi.deps.vendor.cpp(it) line.  However, my libraries are part of the sources.cpp block.  Therefore I cannot figure out how to change the order of the vendor libraries in general versus my libraries.  I even tried moving the wpi.deps.vendor.cpp(it) above the sources block but this did not help. 

  
      frcUserProgram(NativeExecutableSpec) {
            targetPlatform wpi.platforms.roborio
			binaries.all {
				cppCompiler.args "-Wall", "-Werror", "-std=c++11", "-g", "-DXEROROBORIO"
			}            
            if (includeDesktopSupport) {
                targetPlatform wpi.platforms.desktop
            }

            sources.cpp {
                source {
                    srcDir 'phoenix/src/main/cpp'
                    include '**/*.cpp'
                }
                exportedHeaders {
                    srcDir 'phoenix/src/main/cpp'
                    include '**/*.h'
                }
				lib library: "xerobase", linkage: "static"
				lib library: "xeromath", linkage: "static"
				lib library: "xeromisc", linkage: "static"                                
            }

            // Defining my dependencies. In this case, WPILib (+ friends), and vendor libraries.
            useLibrary(it, "wpilib")
            wpi.deps.vendor.cpp(it)
        }

ThadHouse, 2019-01-05:
Try putting your libraries in a `binaries.all` block, instead of sources. I'm not sure how sources ordering works relative to binaries.all, which is what ours uses.

sjcbulldog, 2019-01-05:
Worked perfectly.  Sorry to be such a bother. I really just wanted to get all this figured out before the big day tomorrow :).

closed 2019-01-05 by sjcbulldog

ThadHouse, 2019-01-05:
No worries. It makes it much easier that you're providing useful debugging information, and not just going "it doesn't work, fix it". Makes it's much easier to debug and fix.

sjcbulldog, 2019-01-05:
I write software for a living. I work for cypress semiconductor and am usually at the other end of this transaction :)