wpilibsuite/GradleRIO #286: Allow disabling `disableCacheCleanup`

kenzierocks, 2019-01-05:
I prefer to have Gradle clean up my caches automatically, but it seems that GradleRIO overwrites my user's properties to disable this. Can this behaviour be put behind a flag of some sort so I can avoid having this plugin overwrite my preferred configuration? I would like to accept the risk associated with it, mostly because I believe that Gradle _won't_ delete recently used dependencies and I'll be running this every day.

ThadHouse, 2019-01-05:
We could add a flag, but the default will definitely be disable that deletion, so you would have to enable that flag in every project. In our scenarios, cache deletion is dumb and would cause massive confusion, so it will always be our default.

kenzierocks, 2019-01-05:
Could you not just pick up `project.property("<something>.disable.cache.cleanup")` and use that? That would also pick up anything from the user-wide Gradle properties.

ThadHouse, 2019-01-05:
We can't risk a user not having that, getting to a competition, and having their caches deleted and not being able to deploy code. If that were to happen to even 1 team, we will have failed because we didn't do everything we could to solve the problem. The default in GradleRIO will be to automatically disable cache deletion, but we can add a project setting to disable that auto set behavior.

ThadHouse, 2019-01-05:
We were very vocally against gradle having that behavior, and still are. Not all users always have internet, and a build system should easily support that behavior. So for GradleRIO, our default will be to disable the cache deletion.

kenzierocks, 2019-01-05:
I feel the same way, and I don't wish to change that. Perhaps I missed that in the wording, but I wish to have a property that I must _set_ in my _user-wide_ properties (a la `~/.gradle/gradle.properties`) to _disable_ `disableCacheCleanup()`. The code can be something like this:
```
// Only turn off if property is explicitly set to `false`
if (project.findProperty("xxxx.disable.cache.cleanup") != "false") {
    disableCacheCleanup()
}
```

I think this is fairly simple, and in-line with what you want.

ThadHouse, 2019-01-05:
Ah ok. Yeah I'd be ok with that.

ThadHouse, 2019-01-05:
@kenzierocks #287 does it. You'll get a 4 line warning by having that property set, but I'm not going to add a way to disable that warning. Its just too risky if someone accidentally sets that.

kenzierocks, 2019-01-05:
Makes sense. Thank you!

PR #287 (2019-01-06): Add way to disable the force no cache deletion behavior
Fixes #286

The displays a large warning when the property is set, with no way to disable the message. We really do not want teams to have that property enabled.
commit 8da9f530 (2019-01-06): Add way to disable the force no cache deletion behavior (#287)

* Add way to disable the force no cache deletion behavior

Fixes #286

The displays a large warning when the property is set, with no way to disable the message. We really do not want teams to have that property enabled.

* Remove setting examples
closed 2019-01-06 by JacisNonsense