wpilibsuite/GradleRIO #303: Cannot deploy to RoboRIO 2019_v12

minglethepringle, 2019-01-16:
With WPILib 2019.2.1 and RoboRIO on 2019_v12, I'm trying to deploy my code from Visual Studio Code. Running gives me this error:

`RoboRIO Image invalid! RoboRIO: 2019_v12, allowed: [2018_v16, 2018_v17, 2018_v18]`

Here's the full log: https://pastebin.com/YAeH2Z5b

I feel like it has to do something with the "NOTE: You are using an ALPHA version of GradleRIO, designed for the 2019 Season! This release uses the 2018 Core Libraries, however all tooling (GradleRIO + IDE support) is incubating for 2019" message, but I don't know.

Kevin-OConnor, 2019-01-16:
It definitely does have to do with that. It sounds like your project was created with the Alpha. These projects are not portable to the season release due to breaking changes. You will need to create a new project and copy your code in.

minglethepringle, 2019-01-16:
@Kevin-OConnor Yup, found part of the problem. Was using outdated build.gradle, was using alpha release. Now I have latest build.gradle, but my Gradle version is still at 4.9, and it's complaining about incompatibility. How can I update Gradle to the latest, 5.1.x?

Kevin-OConnor, 2019-01-16:
It's not clear from your comment how you updated the build.gradle. Things will be much easier if you 1) Make sure you have the 2019.2.1 version of the extension installed by clicking on the Extensions tab (it's the icon on the left that looks like a box). 2)Create a new project, using VSCode. 3) Copy your code into the new project.

If this is an imported Eclipse project, simply import it again with the new extension. Manually trying to update the project to the 2019 official releases is not recommended.