wpilibsuite/GradleRIO #304: GradleRIO using wrong Java

minglethepringle, 2019-01-16:
I'm getting an error on build: `Could not target platform: 'Java SE 11' using tool chain: 'JDK 8 (1.8)'`

I told VSCode to "set VS Code Java Home to FRC Home". 

Anyone know why this isn't working? I have Java 1.8 installed on system, but I have "jdk-11.0.1.jdk" folder under my WPILib Mac 2019.2.1 folder.

1) How can I actually set VS Code to use the 1.11 JDK under my WPILib folder?
2) How does the "Set VS Code Java Home to FRC Home" thing actually work?

Thanks!

ThadHouse, 2019-01-16:
The folder actually needs to be put in ~/frc2019. So you should have a ~/frc2019/jdk folder. Once you have that, vscode will pick it up. 

When you have that properly extracted, the vscode command will set the vscode java.home variable to point to that location. That is needed to make intellisense work correctly. Builds will work without them, but intellisense might not.

minglethepringle, 2019-01-16:
I put it in ~/frc2019/jdk, but it still didn't work.

I manually set java.home in VSCode settings to the path of the JDK, but it's saying it isn't a valid JDK, and I notice it's missing a Home directory under Contents/MacOS where all the actual JDK stuff is. I'm gonna download JDK 1.11.2 and see if that works instead.

minglethepringle, 2019-01-16:
That didn't work either.

ThadHouse, 2019-01-16:
When you set the java.home variable manually, what did you set it to? You have to expand the ~ manually, it won't expand it for you.

minglethepringle, 2019-01-16:
/Users/mingle_li/frc2019/jdk/jdk-11.0.2.jdk

ThadHouse, 2019-01-16:
No, you just want to set it to 

Users/mingle_li/frc2019/jdk

You don't want the inner folder. We remove the contents/macos/home directories in order to match the other platforms.

minglethepringle, 2019-01-16:
Currently, have jdk-11.0.1.jdk under frc2019/jdk/, with `"java.home": "/Users/mingle_li/frc2019/jdk"` in visual studio's setting.json, still complains.

minglethepringle, 2019-01-17:
Found a workaround. I don't know why it didn't pick up the JDK in the ~/frc2019/jdk folder, but what I did was:

1. Install JDK 1.11.2 system-wide (instead of using the one packaged with WPILib)
2. Set and export my JAVA_HOME to point to the JDK home
3. Do "Set VS Code Java Home to FRC Home" again.

closed 2019-01-17 by minglethepringle