wpilibsuite/GradleRIO #308: SimulateJava and Driver Station

Bankst, 2019-01-31:
I'm working on some code for a custom operator controller that doesn't need to interact with anything but the controller itself, and borrowing a Rio from the team isn't really an option, nor convenient. Is it possible to connect the official Driver Station to the simulated robot? 

I've tried running the DS locally with the simulated bot, and it doesn't see a robot controller. I also tried with a 2nd computer running the DS, with the PC running the RC set to a static IP of 10.TE.AM.2, and the DS on 10.TE.AM.21. This *does* show the robot controller is connected in the diagnostics menu, but still shows "No Robot Communication".

JacisNonsense, 2019-02-10:
You need to add the correct halsim module. Check here: https://www.chiefdelphi.com/t/wpilib-rocks/339145/13?u=jaci

Bankst, 2019-02-11:
I've added the shim to the 'libs' folder in project root, and the platform-check block in 'depencies'.
```
dependencies {
    compile wpi.deps.wpilib()
    compile wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.roborio)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)
    testCompile 'junit:junit:4.12'

    if (wpi.platforms.desktop == 'windowsx86-64') {
        simulation rootProject.files('libs/halsim_ds_socketd.dll')
    } else if (wpi.platforms.desktop == 'osxx86-64') {
        simulation rootProject.files('libs/libhalsim_ds_socketd.dylib')
    } else if (wpi.platforms.desktop == 'linuxx86-64') {
        simulation rootProject.files('libs/libhalsim_ds_socketd.so')
    } else {
        simulation "edu.wpi.first.halsim:halsim_ds_socket:${wpi.wpilibVersion}:${wpi.platforms.desktop}debug@zip"
    }
}
```

I do not get a message "Driver station initialized" or the likes from the robot program, and the DS still does not see the RC.

Bankst, 2019-02-28:
Solution: 
```
dependencies {
    compile wpi.deps.wpilib()
    compile wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.roborio)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)
    testCompile 'junit:junit:4.12'

    simulation "edu.wpi.first.halsim:halsim_ds_socket:${wpi.wpilibVersion}:${wpi.platforms.desktop}@zip"
    simulation "edu.wpi.first.halsim:halsim_ds_socket:${wpi.wpilibVersion}:${wpi.platforms.desktop}debug@zip"
}

Confirmed working with the Dashboard using GradleRIO 2019.2.1 and 2019.4.1

closed 2019-02-28 by Bankst

ThadHouse, 2019-02-28:
Oh. You might just have figured out why it wasn't working. Did the original example have the debug at the end? With java, only debug would work. Can you try removing the non debug one and seeing if it works?

Bankst, 2019-02-28:
Without this line
`simulation "edu.wpi.first.halsim:halsim_ds_socket:${wpi.wpilibVersion}:${wpi.platforms.desktop}debug@zip"`
it does work. Without the non-debug line, it does not work.
Side note, this is without the .dll in the `libs` folder of the project.