wpilibsuite/GradleRIO #311: Simulate runs with project directory as working directory - `src/main/deploy` isn't handled for simulation

JacisNonsense, 2019-02-09:


blu28, 2019-03-04:
I just hit this one. Any ideas for a workaround?

JacisNonsense, 2019-03-04:
You can check if you're in a simulation through `RobotBase.isSimulation`, and add `src/main/deploy` to the path if so

blu28, 2019-03-05:
> You can check if you're in a simulation through `RobotBase.isSimulation`, and add `src/main/deploy` to the path if so

Yes, thank you, I hit on this solution too. Github doesn't seem to have a way to "vote" or say "I hit this one too", but i wanted to make sure you knew it was being seen "in the wild." 

Thank you for all the work you are doing on this. Desktop simulation is a game changer for our team.

Starlight220, 2020-09-17:
[`Filesystem.getDeployDirectory()`](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/Filesystem.java#L59) correctly returns `src/main/deploy` in sim, and files can be retrieved correctly (I just verified this). Therefore I think that this issue has been resolved and can be closed.

AustinShalit, 2020-10-13:
Closed by https://github.com/wpilibsuite/allwpilib/pull/2293/

closed 2020-10-13 by AustinShalit