wpilibsuite/GradleRIO #315: Vendor Library not Deploying Properly - ClassNotFoundException

sleepisfortheweak, 2019-02-14:
Hopefully I'm just missing something here. I'm publishing my vendor library using the JSON method. I've got VS Code picking up the Vendor Library and compiling successdully. However, when the code runs I get a ClassNotFoundException when it attempts to load my class. See attached error, JSON, and JAR files. Thanks for your help.

`
********** Robot program starting **********
Unhandled exception instantiating robot jdk.internal.loader.BuiltinClassLoader java.lang.ClassNotFoundException: edu.wpi.first.wpilibj.util.UncleanStatusException
Error at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source): Unhandled exception instantiating robot jdk.internal.loader.BuiltinClassLoader java.lang.ClassNotFoundException: edu.wpi.first.wpilibj.util.UncleanStatusException
at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source)
at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(Unknown Source)
at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
at frc.robot.Robot.<clinit>(Robot.java:27)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:221)
at frc.robot.Main.main(Main.java:27)
`

[LightDrive-java-2019.02.13.zip](https://github.com/wpilibsuite/GradleRIO/files/2863795/LightDrive-java-2019.02.13.zip)

[LightDrive.json.txt](https://github.com/wpilibsuite/GradleRIO/files/2863796/LightDrive.json.txt)

PeterJohnson, 2019-02-14:
There is no such class `edu.wpi.first.wpilibj.util.UncleanStatusException`.  Did you mean `edu.wpi.first.hal.util.UncleanStatusException`?  I'm not sure how you successfully compiled.

sleepisfortheweak, 2019-02-14:
That error is copied directly from GradleRIO output. Is that a clue to what happened? I just imaged the RIO with the 2019 FRC update and updated VS Code's GradleRIO extension from alpha release to latest. Once I installed my Vendor Library, VS Code picked up the LightDrive classes and built successfully.

ThadHouse, 2019-02-14:
You would need the libraries recompiled for 2019. 2018 libraries will not work with the 2019 setup. Because some of the classes, especially the low level communication classes, have changed.

sleepisfortheweak, 2019-02-14:
Good call. That's probably it. I thought maybe it was a Java version issue, but didn't think about WPILib portion. I will attempt to recompile that JAR against 2019 WPILib.

sleepisfortheweak, 2019-02-15:
Yes, that was the error. I built with the 2019 WPILib and I can now deploy successfully. Thanks for the help.

closed 2019-02-15 by sleepisfortheweak