wpilibsuite/GradleRIO #325: simulateJava task fails

ThePfaffanater, 2019-03-27:
Upon running SimulateJava it proceeds to just open a blank cmd. If you leave echo on and run it from a terminal it responds with the error: 

> @TP-Laptop MINGW64 ~/Desktop/Creative/Programming/Dev Projects/Java/2019-DriverStation/RobotDashboard/build (dev)
$  ./gradlerio_simulateJava.bat
Picked up _JAVA_OPTIONS: -Xmx512M
Error: Could not find or load main class com.team871.FXMLGraphicalMain
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application

So it is obviously failing and doing nothing resulting in the behavior explained above with the blank CMD terminal just appearing. When looking further inside the SimulateJava.bat I found the 
line `"C:\Program Files\Java\jdk-11.0.1\bin\java.exe" -Djava.library.path="C:\Users\TP-Laptop\Desktop\Creative\Programming\Dev Projects\Java\2019-DriverStation\RobotDashboard\build\tmp\jniExtractDir" -jar "C:\Users\TP-Laptop\Desktop\Creative\Programming\Dev Projects\Java\2019-DriverStation\RobotDashboard\build\libs\RobotDashboard-1.0-SNAPSHOT.jar"`

If I replace `"C:\Program Files\Java\jdk-11.0.1\bin\java.exe - jar"` with just `java -jar` there are no problems and it starts right up when ran from the terminal.

ThadHouse, 2019-03-27:
Why is that running in mingw? Mingw likely isn't handling that full path correctly. I think we only test in CMD and powershell.

ThePfaffanater, 2019-03-27:
I would agree with you if it was not erorring out and doing nothing when it is run through Gradle. I did just run it through Powershell and got the same exact result :(

PeterJohnson, 2019-03-27:
Where is java installed?  If it's not in C:\Program Files\Java\jdk-11.0.1\bin...

ThePfaffanater, 2019-03-27:
That is where it is installed, 'Java.exe' exists in that directory. Maybe this could be something with my SystemVariables?

PeterJohnson, 2020-12-05:
This seems to be an issue specific to your system.  We are closing due to lack of activity on the issue; please reopen if you are still having this issue so we can try to debug further.

closed 2020-12-05 by PeterJohnson