wpilibsuite/GradleRIO #373: Having problems with building

katzuv, 2019-11-08:
Hi,
[This](https://github.com/katzuv/Robot-2018-Revised) is our repository, it includes all the Gradle files including [`build.gradle`](https://github.com/katzuv/Robot-2018-Revised/blob/master/build.gradle).

We get the next error when running `./gradlew build --info`:
```2019-11-08T12:46:09.2471541Z Build Exception: class org.gradle.internal.exceptions.LocationAwareException -> Build file '/home/runner/work/Robot-2018-Revised/Robot-2018-Revised/build.gradle' line: 59
2019-11-08T12:46:09.2472515Z FAILURE: Build failed with an exception.
2019-11-08T12:46:09.2472990Z Could not determine the dependencies of task ':jar'.
2019-11-08T12:46:09.2474268Z 
2019-11-08T12:46:09.2474840Z Build Exception: class org.gradle.api.internal.tasks.TaskDependencyResolveException -> Could not determine the dependencies of task ':jar'.
2019-11-08T12:46:09.2475886Z * Where:
2019-11-08T12:46:09.2476467Z Build Exception: class org.gradle.api.internal.artifacts.ivyservice.DefaultLenientConfiguration$ArtifactResolveException -> Could not resolve all files for configuration ':compileClasspath'.
2019-11-08T12:46:09.2477875Z Build file '/home/runner/work/Robot-2018-Revised/Robot-2018-Revised/build.gradle' line: 59
2019-11-08T12:46:09.2478827Z Build Exception: class org.gradle.internal.resolve.ModuleVersionNotFoundException -> Could not find edu.wpi.first.thirdparty.frc2019.opencv:opencv-java:3.4.4-4.
2019-11-08T12:46:09.2479873Z 
2019-11-08T12:46:09.2479996Z Searched in the following locations:
2019-11-08T12:46:09.2480930Z * What went wrong:
2019-11-08T12:46:09.2481481Z   - file:/home/runner/wpilib/2020/maven/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2482791Z Could not determine the dependencies of task ':jar'.
2019-11-08T12:46:09.2483239Z   - file:/home/runner/wpilib/2020/maven/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2484524Z > Could not resolve all files for configuration ':compileClasspath'.
2019-11-08T12:46:09.2485073Z   - https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2486446Z    > Could not find edu.wpi.first.thirdparty.frc2019.opencv:opencv-java:3.4.4-4.
2019-11-08T12:46:09.2486940Z   - https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2488117Z      Searched in the following locations:
2019-11-08T12:46:09.2489151Z   - http://devsite.ctr-electronics.com/maven/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2490658Z        - file:/home/runner/wpilib/2020/maven/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2491182Z   - http://devsite.ctr-electronics.com/maven/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2492598Z        - file:/home/runner/wpilib/2020/maven/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2493106Z   - https://repo1.maven.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2494713Z        - https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2495215Z   - https://repo1.maven.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2496635Z        - https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2497150Z   - http://wpimirror.imjac.in/m2/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2499041Z        - http://devsite.ctr-electronics.com/maven/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2499758Z   - http://wpimirror.imjac.in/m2/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2501403Z        - http://devsite.ctr-electronics.com/maven/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2502202Z   - https://repo.maven.apache.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2503802Z        - https://repo1.maven.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2504326Z   - https://repo.maven.apache.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2505848Z        - https://repo1.maven.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.2506023Z Required by:
2019-11-08T12:46:09.2507395Z        - http://wpimirror.imjac.in/m2/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.2507814Z     project : > com.kauailabs.navx.frc:navx-java:3.1.344
2019-11-08T12:46:09.3430759Z        - http://wpimirror.imjac.in/m2/release/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.3432157Z        - https://repo.maven.apache.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.pom
2019-11-08T12:46:09.3433782Z        - https://repo.maven.apache.org/maven2/edu/wpi/first/thirdparty/frc2019/opencv/opencv-java/3.4.4-4/opencv-java-3.4.4-4.jar
2019-11-08T12:46:09.3434041Z      Required by:
2019-11-08T12:46:09.3434488Z          project : > com.kauailabs.navx.frc:navx-java:3.1.344
2019-11-08T12:46:09.3434558Z 
2019-11-08T12:46:09.3434668Z * Try:
2019-11-08T12:46:09.3435112Z Run with --stacktrace option to get the stack trace. Run with --debug option to get more log output. Run with --scan to get full insights.
2019-11-08T12:46:09.3435199Z 
2019-11-08T12:46:09.3435335Z * Get more help at https://help.gradle.org
2019-11-08T12:46:09.3435429Z 
2019-11-08T12:46:09.3435540Z BUILD FAILED in 29s
2019-11-08T12:46:09.8042976Z ##[error]Process completed with exit code 1.
2019-11-08T12:46:09.8068823Z Cleaning up orphan processes
2019-11-08T12:46:09.8443876Z Terminate orphan process: pid (2716) (java)```
As mentioned we ran with the `--info` flag so the whole log can be seen [here](https://github.com/katzuv/Robot-2018-Revised/commit/c1a80e93185e8cb5a430297ebdc8935f10c4489e/checks?check_suite_id=301973208).

PeterJohnson, 2019-11-08:
You’re trying to use 2019 vendor dependencies (in particular for NavX) with 2020 wpilib.  That won’t work.

katzuv, 2019-11-09:
Thank you, I changed the version of GradleRIO to `2019.4.1`.
However, now I'm getting the following issue:
```
* Where:
Build file '/home/runner/work/Robot-2018-Revised/Robot-2018-Revised/build.gradle' line: 41

* What went wrong:
A problem occurred evaluating root project 'Robot-2018-Revised'.
> No signature of method: edu.wpi.first.gradlerio.wpi.dependencies.WPIDepsExtension.wpilibJni() is applicable for argument types: (String) values: [linuxathena]
  Possible solutions: wpilibJni(), wpilib(), wpilibJars()
```
Is there anything else I need to change to all the version will match?
Thanks!
(Also, I pointed earlier to another version of the `build.gradle` file. [This](https://github.com/katzuv/Robot-2018-Revised/blob/github-ci/build.gradle) is the correct version, sorry about it.)

ThadHouse, 2019-11-09:
When moving back to 2019.4.1, you need the following dependencies block.

```
dependencies {
    compile wpi.deps.wpilib()
    compile wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.roborio)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)
    testCompile 'junit:junit:4.12'
}
```

You can completely replace the existing one. That was on the list of things changed for 2020.

katzuv, 2019-11-09:
Hey, I'm still getting issues, although I believe I fixed all the version-compatibility issues. Links:
- [`build.gradle`](https://github.com/katzuv/Robot-2018-Revised/blob/github-ci/build.gradle)
- [`gradle.yml`](https://github.com/katzuv/Robot-2018-Revised/blob/github-ci/.github/workflows/gradle.yml)
- [Log](https://github.com/katzuv/Robot-2018-Revised/commit/bbe4222822cae7ea47fb7a3ebca77b535e2e4fca/checks?check_suite_id=303600871)

Thanks a lot for your help!

prateekma, 2019-11-09:
The first.wpi.edu domain seems to be down today

katzuv, 2019-11-10:
Oh, you're probably right. The domain was up for a few minutes, then the build was successful, now the site is down again and the build fails again.

katzuv, 2019-11-10:
Thanks, everyone! Finally, I managed to get the pipeline up and running. Thank you all for your help!

closed 2019-11-10 by katzuv