wpilibsuite/GradleRIO #390: Installer and InstallAllTools tasks do not play nicely together

ThadHouse, 2020-01-03:
The casing is different for shuffleboard, causing them to not be seen as the same executable.

sciencewhiz, 2020-01-04:
VS Code should change to be uppercase.

Starlight220, 2020-10-15:
When mentioning "Installer", is the intention to the WPILib installer or the individual tool install tasks? 
If the former, this issue should be moved to the installer or VS Code repo

Starlight220, 2020-12-05:
#### Current situation
The problem is that the shuffleboard artifact is cased `shuffleboard` while the rest are cased `SmartDashboard`. The installer creates the executable based on the artifact name ([here](https://github.com/wpilibsuite/WPILibInstaller-Avalonia/blob/f5d4f9192764563b1111af5a15d75ba39374326c/apps/ToolsUpdater/src/main/java/Program.java#L37)), meaning that the executable's name is `shuffleboard`. The `tools.json` (that I can't find where it's generated) also has `"name": "shuffleboard"`, so that doesn't help. GradleRIO sets the artifact name [here](https://github.com/wpilibsuite/GradleRIO/blob/adcee14ea6b8ce120e8536cd712a9320f79517da/src/main/groovy/edu/wpi/first/gradlerio/wpi/dependencies/tools/WPIToolsPlugin.groovy#L25) as `Shuffleboard`. 

#### What needs to happen
One of two possibly conflicting things :
- Move the maven location to be cased `Shuffleboard` like the rest of the tools, this move should be combined with moving Shuffleboard to the `tools` namespace. Everything that uses the maven location will need to be updated - this is the solution with more work, but it'll be done anyway at some point. Also, this will change the name to `Shuffleboard` as @sciencewhiz specified.
- Change the naming in GradleRIO ([here](https://github.com/wpilibsuite/GradleRIO/blob/adcee14ea6b8ce120e8536cd712a9320f79517da/src/main/groovy/edu/wpi/first/gradlerio/wpi/dependencies/tools/WPIToolsPlugin.groovy#L25)) to `shuffleboard` - this is a single string change that doesn't break anything. The naming will be `shuffleboard` everywhere.

#### So, what do we do?
Option 2 is a no-brainer change, I can submit a PR for it in a second. Option 1 will need PRs in Shuffleboard, Installer, GradleRIO, and possibly another repository or two; each of those PRs will be changing a single string. Having the naming `Shuffleboard` is more correct and consistent with the other tools.

@ThadHouse, @PeterJohnson - what's the verdict?

ThadHouse, 2020-12-05:
I'm 100% opposed to #1 because it will break on case insensitive file systems like windows. This is why it wasn't fixed when we noticed it years ago. So 2 is the only option.

Starlight220, 2020-12-05:
Issue Status: 
Shuffleboard will move to `edu.wpi.first.tools:Shuffleboard:version` for 2022. Nothing will be done about this before that.

PR #604 (2022-06-19): Update Shuffleboard artifact location
Closes #390 

Would this need a change in the installer as well?
closed 2022-06-19 by PeterJohnson