wpilibsuite/GradleRIO #391: Template Project deployJreRoborio Fails

Chris2fourlaw, 2020-01-05:
Deploy to robot fails. Tested on new timed robot template project with no code changes.

https://pastebin.com/raw/pUsaUU8T

ThadHouse, 2020-01-05:
If you didn't use the offline installer, you need to run a build once before running a deploy, in order to cache that file.

PeterJohnson, 2020-01-05:
You need to use the installer first so the artifacts are local to your computer.

Chris2fourlaw, 2020-01-05:
Building works fine, saw downloads and everything on the first time. Guess this file isn't being downloaded?

ThadHouse, 2020-01-05:
The default deploy in VS Code always deploys in offline mode. I'll look into that, but you can open an integrated terminal with Ctrl + ~, and then just run

`./gradlew deploy  -PteamNumber=9973  -Dorg.gradle.java.home="/Library/Java/JavaVirtualMachines/openjdk-11.0.2.jdk/Contents/Home"`

That will cause a deploy to run without offline mode. You will only need to do this once, and you will need internet.

Chris2fourlaw, 2020-01-05:
That solved it. Thanks. I think it would be beneficial to throw that download in with build if possible.

Starlight220, 2020-11-30:
I believe the missing step here is `installRoborioToolchain`, but I'm not that sure. If it is, adding the `installRoborioToolchain` as a dependency of `deployJreRoborio` would fix this. @ThadHouse, am I missing something here? I can fix this if this is indeed the problem.

PeterJohnson, 2020-11-30:
No, installRoborioToolchain is used to install the C++ compiler on the local system for use with C++ projects.

Starlight220, 2020-11-30:
Then how do we want to resolve this? What's missing is `edu.wpi.first.jdk:roborio-2020:11.0.4u10-2`, I believe that is the RoboRio C++ compiler that is installed using `installRoborioToolchain`.

ThadHouse, 2020-11-30:
Its supposed to grab the jdk during build, but we've seen cases where Gradle just decides to bypass that for no reason. There's not much we can do, I can't consistently repro it.

Starlight220, 2020-11-30:
Ok. So any ideas as to how to fix this? I assume that the jre fetching task is a dependency of `deployJreRoborio`? What task is that?

PeterJohnson, 2020-12-01:
Going back to the original pastebin, this was a C++ project?  Why was it trying to deploy the JRE at all?

ThadHouse, 2020-12-01:
Oh I didn't even notice that. Thats the bug. `build` in C++ won't download that. I'll dig in, because thats something I hadn't seen.

Starlight220, 2020-12-20:
I think I found the issue: 
https://github.com/wpilibsuite/GradleRIO/blob/6646f69cd953c03a1be98c96662a18e1075151a6/src/main/groovy/edu/wpi/first/gradlerio/frc/FRCJREArtifact.groovy#L15
That function/whatever should return `false` if it's a C++ project.

ThadHouse, 2020-12-20:
It's set to false in other places.

ThadHouse, 2022-01-28:
This was fixed in the rewrite, as the JDK deploy is 100% contained in java projects now.

closed 2022-01-28 by ThadHouse