wpilibsuite/GradleRIO #393: Multi module project only builds jars for wpi module

retrodaredevil, 2020-01-10:
I have a unique set up for our project: We have four different gradle subprojects or modules. There's core, wpi, gdx, and desktop. Only the core module is used in the wpi module. All the normal tasks from GradleRIO work perfectly by doing things like `./gradlew wpi:deploy` or `./gradlew wpi:simulateJava`. However, it doesn't actually recompile the core module. This means that if I make changes to the core module, I have to run `./gradlew jar` or just `./gradlew core:jar`. 

I know that core is not being recompiled because I get this error by running `./gradlew clean wpi:simulateJava`:

```
* What went wrong:
Execution failed for task ':wpi:jar'.
> Cannot expand ZIP '/home/josh/Programming/First1444/2020/robot2020/core/build/libs/core.jar' as it does not exist.
```
My current fix for this is running
```
./gradlew jar
./gradlew wpi:<my task>
```

Should this be something I fix in my jar { ... } configuration in the build.gradle in my wpi module, or is this a bug in GradleRIO? I'm sure there's a way to recompile core when compiling wpi, but I figured this might be something other teams have trouble with should they decide to have a unique set up like we do.

Our github: https://github.com/frc1444/robot2020

ThadHouse, 2020-01-10:
I've seen Gradle have issues trying to set subproject dependencies from the root project. Try setting all dependencies from their specific projects.

retrodaredevil, 2020-01-10:
I tried moving `api project(":core")` to the dependencies in wpi/build.gradle and that didn't work. I also tried doing that for all other dependencies in /build.gradle and that did the same thing.

I've had similar problems before with other Gradle projects, but I don't remember how I fixed it. My next solution might be to just try and invoke core's jar task from wpi's jar task and see if that works. But it would be nice if a more elegant solution presented itself.

ThadHouse, 2020-01-10:
The only custom stuff we do with the jar is the jar block in the user file. Otherwise, we just use the jar. The collect might be doing an odd thing with the inter project dependency, I'm not sure. Also, try implementation rather then API.

We don't do anything special with the jar though, so it's something that likely needs to be configured on your projects side

retrodaredevil, 2020-01-10:
I have it fixed now. I needed to add `dependsOn ':core:jar'` to the `jar {...}` in `wpi/build.gradle` for it to work.

Thanks for the help

closed 2020-01-10 by retrodaredevil