wpilibsuite/GradleRIO #394: Fail deploy if natives won't be deployed

PeterJohnson, 2020-01-10:
Several users have had crashes at runtime due to changing the GradleRIO version to 2020 instead of using the upgrade tool.  The crash backtrace doesn't point to the root cause of the problem, so users have to search online / ask questions to understand the required fix.  The root cause of this is that the deploy changed from 2019 to 2020 to need a separate line in build.gradle to deploy the native binaries.  Since deploying without those lines is basically guaranteed to fail at runtime, it would be better to catch this case and error out during the deploy step with a more helpful error message.

Starlight220, 2020-11-18:
Is there a risk that this will happen now in 2021?

PeterJohnson, 2020-11-21:
No.  This did not change from 2020 to 2021 so it’s not a risk to users at this point.  Closing.

closed 2020-11-21 by PeterJohnson