wpilibsuite/GradleRIO #397: Gradle RIO: Install Shuffleboard; Include plugins

oconnelc, 2020-01-20:
We currently have a Shuffleboard plugin that we'd like to include with the install. 

We would like to have our plugin project get built before the Shuffleboard install. I know how to update our build.gradle to change the tasks dependencies to do this but the problem is that the Shuffleboard task doesn't seem to have a way to include plugins in it's dependencies.

ThadHouse, 2020-01-20:
The way to do this is just

ShuffleBoard.dependsOn yourPluginTask

That will make it so your plugin generation task runs before the shuffleboard execution works. Just standard Gradle dependency management.

oconnelc, 2020-01-20:
@ThadHouse I appreciate the help. All that will do is cause our build process to run before the Shuffleboard task. It won't install the plugin in shuffleboard

ThadHouse, 2020-01-20:
Shuffleboard plugins work by copying them into a folder. It's fine to do that as long as that happens before shuffleboard starts. The install task is just copying the jar file to the tools folder, but plugins install to a different folder.

oconnelc, 2020-01-20:
I should clarify my question: I know how to setup gradle for the proper build order, but I do not know where to put the plugins. When I look at how Shuffleboard runs from gradle, all I see is that the jar is copied into: C:\Users\Public\wpilib\2020\tools and then run as a java jar. 

I'm not seeing anywhere to drop the plugin files.

closed 2020-01-20 by oconnelc