wpilibsuite/GradleRIO #399: Task `simulateJava` causes code to crash in examples

Jamdan2, 2020-01-24:
Hi all, it seems like when running the task `simulateJava` in the GradleRIO examples, the code crashes at runtime. This problem is not present when pushing to the RoboRIO. The error is as follows:
```
java.io.IOException: wpiHaljni could not be loaded from path or an embedded resource.
        attempted to load for platform /windows/x86-64/

        at edu.wpi.first.wpiutil.RuntimeLoader.loadLibrary(RuntimeLoader.java:78)
        at edu.wpi.first.hal.JNIWrapper.<clinit>(JNIWrapper.java:38)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:316)
        at frc.team0000.robot.Main.main(Main.java:27)
```
This has not happened when running `simulateJava` in the past. Is this a change that was made or is it a bug?

PeterJohnson, 2020-01-24:
Did you install using the WPILib installer?  If not, you need to install the latest Visual Studio redistributable.  https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads

Jamdan2, 2020-01-26:
Thanks, that was the issue.

PR #401 (2020-02-18): Make Java Simulation task use wpilib JDK, message about potential missing runtime
Closes #399 and #400
closed 2020-02-18 by ThadHouse