wpilibsuite/GradleRIO #409: CLion Support Improvements

Modelmat, 2020-03-19:
The current `./gradlew clion` generates non-working CMakeLists, erroring out with
 ```
CMake Error at CMakeLists.txt:89 (add_custom_target):
  add_custom_target cannot create target "frcUserProgram_release_build"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
```
This can be fixed by manually removing the majority (duplicated) of the CMakeLists.txt file; as [this](https://gist.github.com/Modelmat/bff69ca27936bf334e8db4eb79cd8f74) shows (three files, the full error log, the non-working CMakeList, and the (IntelliSense)-working cmakelist).

However, I have noticed that it is possible to use the `compile_commands.json` file to set up the project; CLion supports this for finding include-paths and building does not work (in fact, the button is greyed out so there is no possibility of an "error" because the CMakeLists does not properly work).

It is easiest to have the `compile_commands.json` stored in the project root, hence I think that having `./gradlew clion` run `./gradlew generateCompileCommands` and then copying the `./build/compile_commands/linuxathenta/compile_commands.json` would work well.

ThadHouse, 2020-04-17:
Can clion be told where to find compile_commands.json? That usually works better, especially since sometimes users want the roborio build, and sometimes they want the desktop build.

Modelmat, 2020-04-17:
As far as I could tell it always seemed to take the folder with `compile_commands` as the project root. Creating a project based off `compile_commands` is also easier in the root.

If it is possible to tell it I don't know how.

ThadHouse, 2020-04-17:
https://www.jetbrains.com/help/clion/compilation-database.html

It looks like you can change the project root, which basically changes the compile_commands location. I would prefer to do that.

Modelmat, 2020-04-17:
Oh. I've somehow missed that. That works fine.

Should this just be removing the CmakeLists file & instructing people to run generateCompileCommands, or should clion somehow be an alias for compile commands generation?

Starlight220, 2020-10-18:
So we just want to remove the `clion` task or change it to be an alias for `generateCompileCommands`? And if so, what would `cleanClion` do?

ThadHouse, 2022-01-28:
Clion support was removed from the rewrite.

closed 2022-01-28 by ThadHouse