wpilibsuite/GradleRIO #432: Allow users to specify a custom vendordeps folder location

Ewpratten, 2020-10-06:
I have run into an issue with many projects, where my team uses multi-project gradle builds for everything. GradleRIO is requiring us to have a copy of our vendordeps folder in every single subproject.

Would someone be able to add a small flag or method somewhere that we can call to make the plugin use a different directory? (For example, we would use `rootProject.file("vendordeps")` instead of `project.file("vendordeps")`)

I am currently stuck with adding a copy task at the very start of the gradle task graph, and relying on it to copy the files around to every subproject.

Starlight220, 2020-10-06:
Theoretically, this can be done by making it a plugin property instead of hardcoded here: https://github.com/wpilibsuite/GradleRIO/blob/7473e314ac61f8ee5a89b027c2f2a7b75da69ee7/src/main/groovy/edu/wpi/first/gradlerio/wpi/dependencies/WPIVendorDepsExtension.groovy#L34
However, I think the profit from this might not be worth the risk of tough debugging in the case a team messes some stuff up. @ThadHouse - your thoughts? If you give the green light, I can work on this (unless @Ewpratten plans to).

ThadHouse, 2020-10-06:
I'm fine with this, as long as a warning is always printed if that value is changed from the default. Adding configuration knobs always makes debugging harder, and I see the use in this but I see how a team could accidentally change it, and then debugging is a nightmare.

Starlight220, 2020-10-06:
I feel that some teams would dislike the constant warning printing. But that can be discussed while a PR is in the works. 

@Ewpratten - do you want to implement this or should I?

Ewpratten, 2020-10-06:
@Starlight220 you can go ahead

ThadHouse, 2020-10-06:
The warning would only print if the value is changed from the default. We do this in other places too. The small number of teams that care about changing the defaults shouldn't have a problem dealing with it. Things that make debugging harder for the majority of teams need to be marked.

PR #433 (2020-10-09): Add `wpi.deps.vendor.folderpath` as mutable property
Resolves #432
 
@ThadHouse - the warning message probably needs some rewording/reworking, any ideas? And I couldn't find other locations of the "default-or-warning" part, feel free to point me in the direction of one to imitate.
closed 2020-10-09 by ThadHouse