wpilibsuite/GradleRIO #437: Versioning should be done on Git tag

Daltz333, 2020-10-09:
Do not hardcore, but use git tags

Starlight220, 2020-10-09:
I think the version might need to be hardcoded (or at least accessible from plugin code) for release. Unless maybe we hardcode to `gradle.properties` (#429 ) and publish in CI.

Starlight220, 2020-10-18:
@ThadHouse do you think that it's realistic/practical to publish via CI that hardcodes the latest git tag to `gradle.properties`?

Starlight220, 2020-11-21:
Hasn't this been handled by #444?

Daltz333, 2020-11-21:
yes

AustinShalit, 2020-11-21:
Not yet. We still need to make a commit to update the version manually.

https://github.com/wpilibsuite/GradleRIO/blob/master/gradle.properties#L1

Starlight220, 2020-11-22:
A commit editing `gradle.properties` will be needed anyway. The trick is that we might be able to have the commit done by GH Actions, though then the tag will be on the prior commit. 
We can have a manually-triggered workflow that commits the version change and tags that commit.

prateekma, 2020-11-22:
We can get the version number from the latest tag so bumping the gradle.properties manually or automatically isn't required.