wpilibsuite/GradleRIO #447: Add way to skip binaries for simulateExternalCpp

ThadHouse, 2020-11-01:
Some users want to skip this including the test binaries in order to speed up the build. This isn't too hard.

https://github.com/wpilibsuite/GradleRIO/blob/41f57a4984cc0fb7b404b154441e887ad3fda2bc/src/main/groovy/edu/wpi/first/gradlerio/test/NativeTestPlugin.groovy#L85

Basically here we just need to not make tests always added, but instead make them configurable.

calcmogul, 2020-11-02:
We were doing the following, which builds both tests and desktop:
https://github.com/frc3512/Robot-2020/blob/0cc81f18d7f86e809f39a9c80d0210e97c4c8f06/gradlew-simgui.sh

Now we do this, which skips building the tests:
https://github.com/frc3512/Robot-2020/blob/14d483d4cfd992058ec24b237fd3616d7b56eda6/build.gradle#L143-L146

Starlight220, 2020-12-03:
Would a `skipBinaries` gradle property be enough, or do we want finer configuration?

ThadHouse, 2022-01-28:
This is implicitly fixed in 2022.

closed 2022-01-28 by ThadHouse