wpilibsuite/GradleRIO #480: Add shorter task alias for C++ simulation

calcmogul, 2021-01-05:
The following in build.gradle works, but it would be nice to have built in to GradleRIO.

```gradle
task simulateCpp {
    dependsOn 'simulateFrcUserProgram' + wpi.platforms.desktop.capitalize() + 'DebugExecutable'
}
```

calcmogul, 2021-01-05:
The same could be said for test aliases like this:
```gradle
task test {
    dependsOn 'testRelease'
}

task testDebug {
    dependsOn 'runFrcUserProgramTest' + wpi.platforms.desktop.capitalize() + 'DebugGoogleTestExe'
}

task testRelease {
    dependsOn 'runFrcUserProgramTest' + wpi.platforms.desktop.capitalize() + 'ReleaseGoogleTestExe'
}
```

sciencewhiz, 2022-01-02:
This was resolved by the adition of simulateNative, right?

calcmogul, 2022-01-02:
Only the simulation part. testNativeDebug and testNativeDebug build the tests, but don't run them. There's still no equivalent of the test, testDebug, and testRelease tasks above.

ThadHouse, 2022-01-28:
I renamed this to more fit what the issue is now.

spacey-sooty, 2024-06-05:
Are the C++ tests not run in the build task?

calcmogul, 2024-06-05:
They are, but it builds way too much.

calcmogul, 2026-04-28:
We should also modify VS Code's build button to invoke a single target's build instead of building everything with `gradlew build`.