wpilibsuite/GradleRIO #487: README.md upgrading instructions are borken

ggoebel, 2021-01-21:
The README.md Upgrading instructions tell you to update build.gradle to set the wrapper gradleVersion to 5.0 and then run ./gradlew wrapper

- The gradeVersion needs to be updated to 6.0.
- Overriding built-in tasks deprecated in 4.8 now produces an error. 
```
> Cannot add task 'wrapper' as a task with that name already exists.
```
- see [Configure existing wrapper and init tasks](https://docs.gradle.org/5.2.1/userguide/upgrading_version_4.html#rel4.8:configure_internal_tasks)

I am not a java native or particularly intimate with gradle. but it looks like the existing instructions are for gradle version <=4.7.  example should add example where currently used gradle version is >4.7:
```
wrapper {
    gradleVersion = '6.0'
}
```

ggoebel, 2021-01-21:
looks like the version should actually be 6.0.1

PeterJohnson, 2021-11-05:
Fixed in #488.

closed 2021-11-05 by PeterJohnson