wpilibsuite/GradleRIO #490: Tool run tasks don't run on Linux

ThadHouse, 2021-01-25:
We had a report that the tool run tasks don't run on linux. These definitely should, but its possible something is broken, potentially with differences in casing between the installer and gradlerio with case sensitive file systems.

calcmogul, 2021-01-25:
`./gradlew PathWeaver` works on my machine (Arch Linux). I have WPILib 2021.2.1 installed.

ThadHouse, 2021-01-25:
It was shuffleboard specifically that had the casing issue. Can you try that?

calcmogul, 2021-01-25:
That one works too; version number says 2021.2.1.

Starlight220, 2021-01-26:
Any problem with shuffleboard is likely caused by #390. Is your shuffleboard installed from the installer or the gradle task? Because one of them won't work.
@calcmogul - can you check your tool folder? I suspect that GradleRIO installed a second executable for Shuffleboard. The gradle task looks for `Shuffleboard.exe` while the installer installs `shuffleboard.exe` - meaning that on case-insensitive filesystems GradleRIO won't see the executable from the installer and will run the install task.

calcmogul, 2021-01-26:
Yes, `./gradlew shuffleboard` installs a `Shuffleboard.py` into `~/wpilib/2021/tools` in addition to the `shuffleboard.py` from the installer. Both Python scripts successfully open shuffleboard. Note that I'm on a case sensitive filesystem.

Starlight220, 2021-01-26:
Then this issue is a symptom of 390, meaning that there isn't much to do about it right now. @ThadHouse - do you agree that this can be closed?

Starlight220, 2021-01-31:
Upon [another report on CD](https://www.chiefdelphi.com/t/failed-to-start-tool-pathweaver/390978), it might not be a GradleRIO problem.

ThadHouse, 2022-01-28:
Since this was completely rewritten, and hasn't been reported again, going to close this issue.

closed 2022-01-28 by ThadHouse