wpilibsuite/GradleRIO #497: Discuss is we should run tests on deploy by default

ThadHouse, 2021-02-19:
At least at some point, running a deploy would also run tests. We have a flag in vscode to skip this behavior, and it's documented. I'm not actually sure that's the case anymore. We should discuss what behavior we want for this so we can make it right in 2022

Starlight220, 2021-02-19:
`test`s would run on `deploy`  if `deploy` depended on `build`. Since it doesn't, then tests would never run on deploy.

ThadHouse, 2021-02-19:
Is that the behavior we want? That's what this issue if for, so we can get the behavior we want.

Starlight220, 2021-02-19:
Since many teams rarely simply `build` but usually only deploy, the tests will almost never be run if they don't run on deploy. On the other hand, running tests could fail/delay a critical deploy at an event or something. 
Either way, the configure option in VS Code should work as well as a way to set without vscode (ie a gradle property or whatever). The point of discussion is what the default should be IMO.

I suggest opening this up for community discussion, perhaps on CD or something.