wpilibsuite/GradleRIO #498: Gradle does not build with new java versions

indexlongrun, 2021-02-20:
The gradle build fails with exception when using new java versions (JDK 14,15): 
java.lang.NoClassDefFoundError: Could not initialize class org.codehaus.groovy.vmplugin.v7.Java7  and java.lang.NoClassDefFoundError: Could not initialize class org.codehaus.groovy.reflection.ReflectionCache

What fixed it for me was: 
https://stackoverflow.com/questions/61289461/java-lang-noclassdeffounderror-could-not-initialize-class-org-codehaus-groovy-v

I changed gradle version in
main/examples/java/gradle/wrapper/gradle-wrapper.properties

from:
distributionUrl=https\://services.gradle.org/distributions/gradle-6.0.1-bin.zip

to:
distributionUrl=https\://services.gradle.org/distributions/gradle-6.8.2-bin.zip

And then the build succeeded

ThadHouse, 2021-02-20:
We do not support running on JDKs newer then 11 and Gradle versions other than 6.0 currently. Java sometimes works, but the native setup does break. 2022 we will be updating to the latest version of Gradle, however we still recommend running on Java 11 even then, as that is what we will support on the roborio.

closed 2021-02-20 by ThadHouse

Starlight220, 2021-02-20:
@Daltz333 I suggest reopening this in frc-docs to add to the Known Issues section.

ThadHouse, 2021-02-20:
This isn't really a great thing for known issues, as teams should be using the installer. Something small saying only Java 11 and Gradle 6 is probably fine, but that'd be it. Workarounds of updating Gradle versions are definitely a no.

Starlight220, 2021-02-20:
I meant without the workaround, something like this:

- **Error:** ```java.lang.NoClassDefFoundError: Could not initialize class org.codehaus.groovy.vmplugin.v7.Java7 and java.lang.NoClassDefFoundError: Could not initialize class org.codehaus.groovy.reflection.ReflectionCache```
- **Solution?:** You are somehow using a newer JDK or Gradle version. The only supported versions are Java 11 and Gradle 6.0. If you are using those versions and are getting this error, please [open an issue] (gradlerio/issues/new).

Daltz333, 2021-02-20:
That issue isn't just a JDK version mismatch. It can be a corrupted cache too.

Starlight220, 2021-02-20:
Either way, do we want to doc it as a known issue or something?

Daltz333, 2021-02-20:
I wouldn't. It's so incredibly rare, and it's very generic, not specific to FRC. We should only document it if

1. More than a miniscule minority face it.
2. It's specific to FRC