wpilibsuite/GradleRIO #503: Allow automated versioning of Git-hosted code

TheMageKing, 2021-04-02:
It would be great if, for projects hosted in Git, GradleRIO allowed teams to include the git version in a part of their robot code, so that a robot could print what version of their code it is running whenever it starts up.

I will start working on a Pull Request to accomplish this, if nobody objects to it.

ThadHouse, 2021-04-02:
This is actually something that can completely be done without being built into gradlerio. We would prefer something like this in the documentation (Which I think we have an open issue for anyway), rather then something built into the plugin itself.

TheMageKing, 2021-04-02:
Can it? I think it would involve modifying the gradle `build` task, which is part of the plugin.  Am I missing something?  I fully confess I am not an expert in Gradle.

ThadHouse, 2021-04-02:
All that it requires is a task that generates a file containing the git commit, and then making the jar task see that as source and making compileJava depend on that task. There are lots of examples how to do things like that both on gradles docs and all over the internet.

TheMageKing, 2021-04-04:
Understood, retitling this issue to talk about documentation (which isn't a part of other GradleRIO issues, as far as I can tell)

ctorek, 2021-11-26:
I am looking into taking up this issue and adding the relevant documentation as a page in the frc-docs [Advanced GradleRIO](https://docs.wpilib.org/en/stable/docs/software/advanced-gradlerio/index.html) section. Would the preferred way to do this be to write the file to the `src/main/deploy` directory of the robot project? @ThadHouse

ThadHouse, 2021-11-26:
That's probably the best option, as it will work for both c++ and Java.

ThadHouse, 2022-01-28:
Since this got merged into docs, we will close this issue.

closed 2022-01-28 by ThadHouse