wpilibsuite/GradleRIO #504: Gradle 7 compatibility

msoucy, 2021-04-11:
Attempting to build a robot project using Gradle 7 appears to have [a couple of warnings](https://scans.gradle.com/s/ypsibhfloei3e/deprecations?expanded=WyI0IiwiMTUiXQ), triggered by `DefaultDomainObjectSet` usage. This prevents builds from being started.

ThadHouse, 2021-04-11:
During years, we don't support updating gradle versions. In fact anything newer then gradle 6.0 doesnt work correctly. 2022 will support and require gradle 7.

closed 2021-04-11 by ThadHouse

msoucy, 2021-04-11:
Then it might be a good idea to leave this report open so you can track it for the 2022 branch, instead of closing it and getting more reports from people who try to update.

ThadHouse, 2021-04-11:
We've never supported mid year updates of gradle. Basically every year that has been the case. Most teams just don't update manually, and we never automatically update teams projects in vscode to newer versions. Gradle 7 is no different then gradle 6 was in 2019.

msoucy, 2021-04-11:
I'm not asking for immediately. I'm saying that there are changes that you'll have to make eventually, and letting you know what they are. I figured I'd give you a bunch of advance warning so it doesn't catch you by surprise, but apparently you didn't want the notice.

ThadHouse, 2021-04-11:
We do appreciate the notice, and we're happy to get the reports. There's just no reason to leave the issue open when it's already been fixed in the 2022 rewrite. So thats why it's closed, it's already fixed.

msoucy, 2021-04-11:
That wasn't clear in your message that closed this. "2022 will support 7" could imply either "it's already fixed" or "we'll fix it".

In this case, I'm fine with it being closed, now that that's been made explicitly clear.