wpilibsuite/GradleRIO #541: Import Vendor Dependency From Library

msoucy, 2021-11-04:
Currently, if code is built as a library but pulls in vendor dependencies via vendordeps and json, the exact same json files need to be included in the separate robot code repository. If not, then Gradle will complain about not being able to find some of the dependencies, because using `api()` adds references to the packages, but not the repositories that they come from.

Ideally those could be propagated with the library information, but that seems like it'd be more of a Gradle change itself.

ThadHouse, 2021-11-04:
That's not something we can solve from our side. The maven metadata format has no way to also specify a repository. This would require some changes far below our level.

ThadHouse, 2023-12-12:
Since this isn't something we can do, I'm going to close this.

closed 2023-12-12 by ThadHouse