wpilibsuite/GradleRIO #549: simulateJava does not work on Mac

PeterJohnson, 2021-11-14:
Fails with the following error:
```
> Task :simulateJavaRelease FAILED
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
2021-11-13 16:15:04.862 java[1729:25379] *** Assertion failure in +[NSUndoManager _endTopLevelGroupings], /AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/Foundation/Foundation-1677.201/Foundation/Misc.subproj/NSUndoManager.m:363
2021-11-13 16:15:04.864 java[1729:25379] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: '+[NSUndoManager(NSInternal) _endTopLevelGroupings] is only safe to invoke on the main thread.'
*** First throw call stack:
(
        0   CoreFoundation                      0x00007fff3367e727 __exceptionPreprocess + 250
        1   libobjc.A.dylib                     0x00007fff6c57fa9e objc_exception_throw + 48
        2   CoreFoundation                      0x00007fff336a7a40 +[NSException raise:format:arguments:] + 88
        3   Foundation                          0x00007fff35dbbead -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 191
        4   Foundation                          0x00007fff35cf85fe +[NSUndoManager(NSPrivate) _endTopLevelGroupings] + 440
        5   AppKit                              0x00007fff3085165c -[NSApplication run] + 864
        6   libhalsim_gui.dylib                 0x0000000130b8beb1 _glfwPlatformCreateWindow + 81
        7   libhalsim_gui.dylib                 0x000000013094406f glfwCreateWindow + 495
        8   libhalsim_gui.dylib                 0x0000000130929286 _ZN3wpi3gui10InitializeEPKcii + 934
        9   libhalsim_gui.dylib                 0x00000001308e281b HALSIM_InitExtension + 1259
        10  libwpiHal.dylib                     0x0000000130565cb2 HAL_LoadOneExtension + 1202
        11  libwpiHal.dylib                     0x0000000130565efa HAL_LoadExtensions + 250
        12  libwpiHal.dylib                     0x00000001305468d3 HAL_Initialize + 131
        13  libwpiHaljni.dylib                  0x000000013049f13d Java_edu_wpi_first_hal_HAL_initialize + 13
        14  ???                                 0x000000011d1542b0 0x0 + 4782899888
)
libc++abi.dylib: terminating with uncaught exception of type NSException

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':simulateJavaRelease'.
> Process 'command '/Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home/bin/java'' finished with non-zero exit value 134
```

It needs a JVM arg `-XstartOnFirstThread` to be set on the launched java.

sciencewhiz, 2021-11-14:
Here's the command that it runs now. 

```/Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home/bin/java -Djava.library.path=/Users/josephross/Documents/Robotics/2022beta2Gearsbot/build/jni/release -Dfile.encoding=UTF-8 -Duser.country=US -Duser.language=en -Duser.variant -jar /Users/josephross/Documents/Robotics/2022beta2Gearsbot/build/libs/2022beta2Gearsbot.jar -XstartOnFirstThread
```

PeterJohnson, 2021-11-14:
It seems like https://github.com/wpilibsuite/GradleRIO/blob/431fd11f8a6518dddda79b407fda0a2e5c9ebf9d/src/main/java/edu/wpi/first/gradlerio/wpi/java/WPIJavaExtension.java#L120
Should be editing jvmArgs instead of args.

PR #550 (2021-11-14): Fix application of -XstartOnFirstThread
Fixes #549
commit 545d2c74 (2021-11-14): Fix application of -XstartOnFirstThread (#550)

Fixes #549
closed 2021-11-14 by PeterJohnson