wpilibsuite/GradleRIO #574: Vendors with HTTP link cause insecure protocol error in gradle 7+ on GradleRIO Beta-3

CJBuchel, 2021-12-16:
* Build Platform *
- OS: POP (Linuxx86-64)
- GradleRIO: 2022.1.1-beta-3

* Issue *
Implementing vendors with http url causes a failure to build or download specified vendor due to insecure protocol.
E.g, with rev robotics spark max 2021 .json url: https://www.revrobotics.com/content/sw/max/sdk/REVRobotics.json
I've tried temporarily marking `repositories.all` as `allowInsecureProtocol = true` in `settings.gradle` frc user side. But to no avail. I assume because the links are managed gradlerio side and are not marked to allow the insecure protocol.

* Output *
```
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':linkFrcUserProgramLinuxx86-64ReleaseExecutable'.
> Could not resolve all dependencies for configuration ':frcUserProgramlinuxx86-64ReleaseExecutable_uberlink'.
   > Using insecure protocols with repositories, without explicit opt-in, is unsupported. Switch Maven repository 'WPI3f48eb8c-50fe-43a6-9cb7-44c86353c4cb_0Release(http://www.revrobotics.com/content/sw/max/sdk/maven/)' to redirect to a secure protocol (like HTTPS) or allow insecure protocols. See https://docs.gradle.org/7.2/dsl/org.gradle.api.artifacts.repositories.UrlArtifactRepository.html#org.gradle.api.artifacts.repositories.UrlArtifactRepository:allowInsecureProtocol for more details. 
```

CJBuchel, 2021-12-16:
Post correction.

I'm aware this is an intentional design due to security.
It just poses some difficulty while using the newer gradlerio version with vendors which are still switching over from http to https (like rev).

And there doesn't seem to be many quick fixes until they do fix the problem

ThadHouse, 2021-12-16:
This will be a requirement for all 2022 vendor deps. Revs 2022 files, which you can get off of teamforge, uses https.

2021 vendor deps don't work with 2022 gradlerio anyway, so all vendors have to rebuild their code anyway. Prior files just will not work anyway, irregardless of this issue.

closed 2021-12-16 by ThadHouse