wpilibsuite/GradleRIO #581: Visual Studio Build Tools

modelmat, 2022-01-12:
So far been unsuccessful in installing the build tools to get C++ sim working on Windows.

The Windows 10 SDK is necessary (unlike I was informed) and with that message you get this warning:
```
> No tool chain is available to build for platform 'windowsx86-64':
       - Tool chain 'visualCpp' (Visual Studio):
           - Could not locate a Visual Studio installation, using the command line tool, Windows registry or system path.
```

According to https://github.com/gradle/gradle-native/issues/617 if you specify https://docs.gradle.org/current/userguide/native_software.html#native_binaries:tool_chain `installDir` as `"C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools"` it should be able to build with it, but I've no idea how to do this with how GradleRIO is setup.
How would you do this?

(I'm also not entirely sure why that issue hasn't been fixed, as adding `-products *` to the vswhere command seems to just... work)

ThadHouse, 2022-01-12:
Theoretically, you can literally copy the code in that example and it should work (minus the clang and gcc lines.) But I've ran build tools before so unsure what the issue is.

modelmat, 2022-01-12:
Oh, that does work. Probably should've just tried it.

Just added

```gradle
    toolChains {
        visualCpp(VisualCpp) {
            installDir "C:/Program Files (x86)/Microsoft Visual Studio/2022/BuildTools"
        }
    }
}
```
at the end of the file

closed 2022-01-21 by PeterJohnson