wpilibsuite/GradleRIO #582: frc2 directory header files not found

nickquednow, 2022-01-15:
In our code we are using C++ and for an example of code:
```cpp
#include <frc2/command/SubsystemBase.h>

class ExampleSubsystem : public frc2::SubsystemBase {

 public:

  ExampleSubsystem();

  void Periodic() override;

  void SimulationPeriodic() override;

};
```

we end up getting the error: `fatal error: frc2/command/SubsystemBase.h: No such file or directory`

I did find that Gradle had downloaded the `frc2/*` directory, but I do not think Gradle added it to the system include directory (or at least one of the search paths).

I do not know if this is related, but we did have the 2020 WPIlib Vscode with the 2021 WPIlib extension installed previously, and installed the 2022 WPIlib VSCode on the same system.

sciencewhiz, 2022-01-15:
Did you include the new command vendordep?

nickquednow, 2022-01-15:
Where would I find it (it was on a brand new installation, and I did not realize that we needed something on a new install)?

boxofrox, 2022-01-16:
@sciencewhiz Looking at our project, here's what I find under the `vendordeps` folder.

```
$ tree robot-code/vendordeps
vendordeps
└── WPILibOldCommands.json

0 directories, 1 file
```

I created a new robot project and checked the `vendordeps` folder of it.

```
$ tree test/vendordeps
vendordeps
└── WPILibNewCommands.json

0 directories, 1 file
```

We did create our robot project with a previous version of the WPILib extension for VSCode by mistake and currently lack `vendordeps/WPILibNewCommands.json`, but this wasn't noticed as we figured we could import our project with the 2022 WPILib extension.  We'll start again and see if that clears up the issue.

boxofrox, 2022-01-16:
After a bit of testing, it appears that missing vendordep was the source of our problems.  Thanks @sciencewhiz for lending us your expertise.

closed 2022-01-16 by nickquednow