wpilibsuite/GradleRIO #584: roboRIO 2 Deploy Hangs at 80%

JCTKO, 2022-01-17:
gradleRIO v2022.2.1
Gradle v7.3.3
roboRIO 2 Image 2022.3.0
Java 11

When running `.\gradlew deploy`, the deploy hangs at 80%. We freshly flashed the roboRIO and the repo is a newly created command bot template. This is the code we built and are trying to deploy: https://github.com/jmskeoch/test-rio2.

It hangs and terminal output is as follows:
```
Task : deployprogramStartfrcJavaroborio:

-C->./etc/profile.d/natinst-path.sh; /usr/local/frc/bin/frcKillRobot.sh -t -r 2> /dev/null @/home/lvuser
```

with debug on this keeps repeating:
[image]

ThadHouse, 2022-01-17:
Have you tried a different computer to deploy from? It's very weird that it would get that far and fail. I wonder if there's a firewall issue somewhere.

sciencewhiz, 2022-01-17:
Can you ssh to the roboRIO and run the command and see what the output is?

JCTKO, 2022-01-17:
We have been deploying from three separate computers, though I can't guarantee that all firewalls were off. It certainly would not be ideal if we needed to keep our firewalls off every time we wanted to deploy.

We have run the command successfully, or at least without error messages and without hanging, by ssh as admin. I'm not sure we've run it successfully as lvuser.

We will try both of these things either tomorrow or Wednesday and get back to you.

JCTKO, 2022-01-21:
We managed to deploy after imaging directly to the SD card and using roboRIO Team Number Setter instead of roboRIO Imaging Tool. It is not mentioned anywhere online, and we happened upon it, and chose to use it, by chance. This worked for both of our RIO 2s.
It doesn't seem like it was a firewall issue, and might have been something like using an old version of the Imaging Tool.

Thank you for the prompt responses.

closed 2022-01-21 by JCTKO