wpilibsuite/GradleRIO #585: Unable to build with any 2022 version

Personator11, 2022-01-18:
Attempting to build project using either 2022.1.1 or 2022.2.1 produces an error:
    

> An exception occurred applying plugin request [id: 'edu.wpi.first.GradleRIO', version: '2022.2.1']
> Failed to apply plugin [class 'edu.wpi.first.deployutils.deploy.DeployPlugin']
   > Could not create an instance of type edu.wpi.first.deployutils.deploy.DeployExtension.
      > 'org.gradle.api.ExtensiblePolymorphicDomainObjectContainer org.gradle.api.model.ObjectFactory.polymorphicDomainObjectContainer(java.lang.Class)'

The projects builds successfully when specifying any 2021 version of GradleRIO in build.gradle.

Daltz333, 2022-01-18:
We do not support manually updating your version in build.gradle. There were substantial changes made to the GradleRIO backend that makes this process incompatible. You must update your project using the VSCode project importer.

Personator11, 2022-01-18:
Okay, I tried and the importer did not recognize my build.gradle as a valid project, so I have just made a new project and transferred the source code.

Daltz333, 2022-01-18:
Interesting. I'm going to go ahead and close this issue. If you have further questions, you can feel free to comment.

closed 2022-01-18 by Daltz333

sciencewhiz, 2022-01-18:
Can you create an issue in vscode-wpilib with a link to your code so that the cause of the importer not recognizing your project can be troubleshot?

Personator11, 2022-01-18:
Sure