wpilibsuite/GradleRIO #590: Feature Request: Differentiate between Desktop Simulation and Desktop-with-hardware

JCaporuscio, 2022-02-05:
Up until now the desktop use-case has only been for pure simulation.

With CANivore the Phoenix libraries now support running a desktop application that utilizes a locally (USB) connected CANivore for the CAN bus.
We can currently support this by having users swap out the Phoenix vendordep depending on if they want pure simulation or hardware-attached simulation, but it would be significantly easier if GradleRIO/vscode natively supported each of these configurations.

Ideally this would be an addition to the debug drop-down menu with an appropriate key addition to the vendordep json spec.

ThadHouse, 2022-02-05:
I'm not sure this is something we could get done for this season, as I'd probably want to take some time to flesh out the design. I definitely think its worth doing something though. Getting this working in C++ is also more invasive especially on the VS Code side, as we would have to switch between modes before doing a build, which is a pretty major change.

virtuald, 2022-02-20:
On the RobotPy side, I'm strongly biased towards some kind of runtime switch instead of a compile time switch.

JCaporuscio, 2023-03-02:
I think this is complete and can be closed?  Since "Hardware Simulate Robot" is now an option.
Unless there's still a user-facing documentation component that's still missing.

ThadHouse, 2023-09-01:
This can be closed. I want to do more documentation, but thats just a general thing.

closed 2023-09-01 by ThadHouse