wpilibsuite/GradleRIO #591: RiologPlugin Removed from 2022 version

oconnelc, 2022-02-19:
In previous versions of the GradleRIO plugin there was a plugin called RiologPlugin that provided the task: riolog${target.name.capitalize()}"

This plugin was removed from the 2022 rewrite. I was unable to find any documentation as to why this was removed and was wondering if this capability could be brought back.

Daltz333, 2022-02-19:
This was an intentional change and was unsupported, even when it was implemented. The only features in GradleRIO that are supported are the ones listed here: https://docs.wpilib.org/en/stable/docs/software/advanced-gradlerio/gradlew-tasks.html

oconnelc, 2022-02-19:
Thank you for the link. It looks like there's an equivalent task: 
`./gradlew riolog - Runs a console displaying output from the default RoboRIO (roborio)`

ThadHouse, 2022-02-19:
It was removed because it wasn't well implemented, and during the rewrite it wasn't worth the dev time to reimplement. It wasn't used much, and riolog exists in both VS Code and the DS itself. 

If we were going to write it again, I would prefer to write a standalone tool with ImGui, as that would be much more usable, and easier to maintain for us. (In fact I've kind of already started working on this). I don't see us ever readding the capability to gradlerio.

oconnelc, 2022-02-19:
The link that was posted mentioned it in the supported tasks, but it doesn't seem like it's there.  This explanation makes the most sense, so we'll use one of the other options. Any chance of getting the documentation updated so that it's not listed as a supported operation?

ThadHouse, 2022-02-19:
@Daltz333 we should remove that task from the docs.

Daltz333, 2022-02-19:
Oops. Didn't even notice it there.

oconnelc, 2022-02-19:
Thank you everyone for your help.

ThadHouse, 2023-12-12:
Since this was a won't fix, its getting closed.

closed 2023-12-12 by ThadHouse