wpilibsuite/GradleRIO #627: Warn if WPILib JDK isn't found

Starlight220, 2022-11-24:
(Not sure if this issue should be here or in the installer repo, or both. Transfer if needed)

Using a different JDK can cause all kinds of issues, such as character encoding problems in the Java-based tools. We should warn if the WPILib-installed JDK isn't found, with a link to the installer etc.

rzblue, 2022-11-24:
I don't know if I'd want to warn with GradleRIO, to be honest. I understand the need for it - I just don't think "forcing" it into GradleRIO is the right move.

ThadHouse, 2022-11-24:
Well it would just be a warning. However, GradleRIO isn't a great place for it, as the logic gradlerio uses is different then the logic the tools use. The only way to do this right would be to have the tool scripts check, but thats invasive too. I'm not really sure how to do this best.