wpilibsuite/GradleRIO #655: MalformedInputException in output.txt crashing the whole compilation

Wylwi, 2023-03-29:
I speak for my former team (5553) as an alumnus.

When compiling, we got this strange error: 

```
2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':compileFrcUserProgramWindowsx86-64ReleaseExecutableFrcUserProgramCpp'.
java.nio.charset.MalformedInputException: Input length = 1
```

Looking at the stacktrace, we could identify that the problem is as follows:
-The compiler generates a warning for a given line of code
-This warning lands in the output.txt file of the running task in a Windows-1252 or similar encoding (but not UTF-8 or standard)
-The method printWarningsForBinTask (in the package edu.wpi.first.gradlerio.wpi.cpp, class WPINativeCompileRules) reads the output file to display it
-By reading the file, an encoding error (see above) appears at line 106 and makes the whole compilation crash

It seems that on our side, and on this computer specifically, there is also the use of a bad compiler. 

However, it does not prevent the bad reading of the file from crashing the whole compilation. I think it would be beneficial to handle this exception better by possibly warning the user and/or avoiding crashing the whole compilation **without more details**.

The team being in the rush of a competition, I didn't have the time to get directly the output.txt file in question, time being short.

I remain at your complete disposal for more information if needed

ThadHouse, 2023-03-29:
We likely won't be doing another wpilib release this year, but I can make sure this gets fixed for next seasons. Probably should just catch and ignore this error completely if it occurs.

Wylwi, 2023-03-29:
Whoops, closed it by accident

closed 2023-07-27 by ThadHouse

ThadHouse, 2024-01-21:
@Wylwi It doesn't look like my attempted fix actually fixed the issue 
https://www.chiefdelphi.com/t/error-code-c/451342

Would you be able to get me a full `--stacktract` log, and the output.txt that is erroring, so I can try to test?

Wylwi, 2024-01-21:
Yep, was discussing it with the team today, I got you an output.txt file ! 
[output.txt](https://github.com/wpilibsuite/GradleRIO/files/14002225/output.txt)

It contains some characters with French accents

Wylwi, 2024-01-21:
Just gonna mention @mimi980, the current student tech lead of the team, he is the one experiencing the issue, I'm not. 

We found out there is probably an issue related to the build target, since the output.txt was found in `build/tmp/compileFrc........Windowsx86_64......` which is clearly not the roborio architecture ! But we don't know how to confirm or refute this theory. On my computer, everything works fine and compile for a linux athena target

ThadHouse, 2024-01-21:
I actually figured out why the fix didn't work. Because of the lambda, the string code is actually throwing an UncheckedIOException and not an IOException. So the try catch I added didn't actually catch the exception. #719 now just catches all exceptions and should fix this for good. Other than you'll always get a log print saying that it couldn't read the file, but we can't easily solve that one.

ThadHouse, 2024-01-21:
Just curious, do you know what codepage your computer is set to @mimi980?

Wylwi, 2024-01-21:
> I actually figured out why the fix didn't work. Because of the lambda, the string code is actually throwing an UncheckedIOException and not an IOException. So the try catch I added didn't actually [...]

Yep, figured it out also. I was doing a PR, but you are too quick! Thanks for your help for this issue!

Wylwi, 2024-01-21:
> Just curious, do you know what codepage your computer is set to @mimi980?

He is running an up-to-date version of Windows 11, 64 bits with WPILib 2024. It looks like Vscode's default is set to UTF-8, but can't say for the whole OS. I think it's a compiler misconfiguration somehow, since I don't understand why warning are thrown in French language...

PR #656 (2023-07-27): Handle MalformedInputException in warning printer
Closes #655 

Its not easy in Java to open a file in an alternate format. So instead, just make an error here a warning and not an error.